jgeumlek/MoltenGamepad #3: Bundling drivers?

TiZ-EX1, 2016-04-22:
Hi there. It looks like you have started to do in C++ what I was going to do with bash and xboxdrv. I'm going to back you as much as I can on this and I'll start with testing this and making drivers, but I am very uncomfortable with C++ so I can't hack on the codebase much, if at all.

So one thing I want to look into is possibly bundling pre-made drivers with the codebase such that users can get their controllers to mimic x360 controllers and work with Unity and SDL2 games out of the box without having to make their own molten or SDL2 mappings. Is this something you'd be interested in doing?

If so, I can get access to a wide variety of controllers due to my proximity to an arcade; expect me to contribute a buttload of fight sticks and adapters. I have a few such controllers in my own possession.

jgeumlek, 2016-04-22:
Yes, I would like to bundle drivers, and that'll be more likely once some internals are cleaned up and a nice ABI forms such that drivers can be compiled separately.

Testing on a variety of input devices is always welcome. I have only a small collection.

Are you aware of the gendev functionality here? What is needed is just a `.cfg` file to describe the device, no C++ required. As long as the device doesn't need fancy features, that should be all it takes. In this case, the matter is just bundling those files.

https://github.com/jgeumlek/MG-Files

I have a separate repo for those (pretty bare right now), and contributions/pull requests to that repo are quite welcome.

TiZ-EX1, 2016-04-22:
I've read the docs; I know I'll just be making .cfg files. I just can't contribute to the internals.

I'll fork that repo and make pull requests for the gendev configs I make. :)

jgeumlek, 2016-04-22:
That sounds great! Sorry about the confusion.

TiZ-EX1, 2016-04-22:
No problem, man.

closed 2016-04-22 by jgeumlek