jgeumlek/MoltenGamepad #6: Some games crash when the virtual keyboard is enabled

nicooo, 2016-05-02:
It seems to happen with GameMaker games

    GAMEPAD: device name Virtual Keyboard (MoltenGamepad)
    ./run.sh: line 3: 28389 Segmentation fault      LD_LIBRARY_PATH=./lib $LD_LIBRARY_PATH ./runner

jgeumlek, 2016-05-02:
Interesting. I have no clue why that would happen.

It only happens when MoltenGamepad is running? Does it not happen when `--no-make-keys` is used?

nicooo, 2016-05-02:
It's fine with `--no-make-keys`.
You can try it with Iron Snout which is free.

jgeumlek, 2016-05-03:
Can confirm: Iron Snout does crash when the virtual keyboard is present, during the period of enumerating devices.

Valgrind appears to show them making a negative `malloc` followed by a null pointer reference. Other devices, virtual or not, appear to be handled fine. It's hard to tell what exactly is going on.

I'll try to investigate some more later to find out the "minimal" virtual device required to make this game crash.

Ryochan7, 2016-05-04:
Based on my testing, the problem stems from BTN_TRIGGER_HAPPY1 (0x2c0) to BTN_TRIGGER_HAPPY40 (0x2e7) being bound to the virtual keyboard. It appears that is enough to make Iron Snout think that the virtual keyboard is a gamepad. Making sure that those keys are omitted is enough for Iron Snout to ignore the virtual keyboard while enumerating devices and launch.

https://github.com/jgeumlek/MoltenGamepad/blob/master/source/uinput.cpp#L145

jgeumlek, 2016-05-04:
Thanks. Removed those as a quick fix, and the game is running fine over here.

It seems adding the mouse axes also leads to Iron Snout running. When mouse emulation finally gets added, we might be able to resume adding TRIGGER_HAPPY (though their usefulness is questionable anyway). KEY_MAX is nice for future-proofing though.

commit 66563579 (2016-05-04): Fix #6 -- exclude BTN_TRIGGER_HAPPY on keyboard
closed 2016-05-04 by jgeumlek