jgeumlek/MoltenGamepad #10: Better slot assignment

jgeumlek, 2016-08-15:
Missing features:

* Easily changing how many output slots are active. For example, for a single-player game, it might make sense to just assign all devices to the single active output slot.
* Automatically assigning a device when added, rather than waiting for the first key press.
* Allowing a device to request a specific slot, e.g. hard-wiring a set of controls to be player one.

TiZ-EX1, 2016-10-21:
I think that since the input that assigns a controller to a slot is still processed, the current behavior is okay, but should be improved; the assignment should be triggered by not only buttons, but also hats and axes pushed beyond a certain threshold (50%, maybe?) I would honestly prefer a controller go into a slot as soon as it's added, but again, I think the current behavior is okay with changes.

jgeumlek, 2016-10-21:
I avoided assignment on axes because of the required threshold logic (otherwise a resting controller's jitter would cause assignment), along with the current inability to tell a trigger from a stick (a trigger's resting value is at the edge of its range!).

Hats are an unfortunate casualty of this. I'll work on getting axes to cause assignments.

Originally assigning-on-connection was avoided since it didn't match my desired behavior when one starts MG while multiple controllers are connected. I wanted to wait for user input to determine which is player one.

There is undocumented support for assignment-on-connection. It is currently controlled by the global setting `auto_assign` in the `slots` category. Check out these commands:

    print options slots
    set slots auto_assign = true

Currently if you want to set this to true at startup, you'd need to put that second line in a profile file loaded at start up, like how I suggested handling the DS4 touchpad. Eventually it will be read from `/mangers/slots.cfg` in the config dir.

If the above works well, it is worth considering changing it to the default behavior. I can't say my octopus-of-controllers set-up is the typical use for MG.

jgeumlek, 2016-10-22:
I just pushed some commits that outdated my comment above:

* Now axis events can cause slot assignment. We roughly check if it has transitioned between being very negative/near zero/very positive. Hats are still problematic. If we receive a very negative value on the very first time we read an axis, is it a hat that was just pushed, or is it a trigger in its idle state? I err on the side of not assigning (at least, not until the user lets go of the hat!).
* The global options such as `set slots auto_assign` are now at least mentioned in the documentation, and the relevant folder has been renamed. For the above, the relevant file is now `options/slots.cfg` instead of `managers/slots.cfg`.

TiZ-EX1, 2016-10-24:
I would personally say an octopus of controllers setup is a perfectly fine usecase, haha.

I tried the auto_assign option, and it works fine.

May I say you've been delightful to work with? I'd love to donate to you to thank you for your time. Maybe there's a Steam game you've been eyeing?

jgeumlek, 2016-10-31:
You've been a pleasure to work with as well. Those suggestions and configs were really helpful.

Thanks for the generous offer, but I'm good for now.

closed 2016-10-31 by jgeumlek