jgeumlek/MoltenGamepad #11: Add Control Socket

jgeumlek, 2016-08-15:
MoltenGamepad acts like a daemon, and its user interface is not very friendly.

A control socket would add a bidirectional way to connect to a running instance of MoltenGamepad. This allows different clients to grow, such as a better CLI or a GUI, without adding extra complexity to the main binary.

There are use-cases for MG without an active X session, and it would be nice to minimize the build requirements.

TiZ-EX1, 2016-10-21:
I think D-Bus is going to be a good target to shoot for mainly because it supports signals; that is, a client can subscribe to the signals and react when a controller is added to show a notification or tweak the LEDs, or maybe when an input device fails driver matching so it knows a GUI should appear for generating a new map.

I don't know what OSC is, but as long as communication is bi-directional, it sounds fine to start with to me.

jgeumlek, 2016-10-21:
I agree that D-Bus would be nice, but I think I'll wait until the feature set and messages are stable.

OSC is just a minimal RPC standard that should suffice. Yes, being bi-directional is key to this. Many languages have OSC libraries.

My projected path for this:

* Standardize how the various locations in MG send output. (Mostly done).
* Build initial version of OSC protocol that is essentially no more than an alternate STDIN/STDOUT stream. This will immediately allow for a `moltengamepadctl` utility that could send commands to a running instance or offer a smoother readline-based user input shell.
* Gradually add more specialized messages to the protocol, like device connections, so that clients aren't parsing STDOUT messages. As we go, add support to `moltengamepadctl` as a reference implementation.
* Eventually write a minimal moltengamepad client library that can connect and maintain information about which devices are connected, so that clients no longer need to deal with the protocol directly.
* Hopefully get some GUI client up and running.
* Eventually transition to D-Bus in the daemon and client library.

jgeumlek, 2017-04-05:
The basic socket protocol has hit the master branch. See moltengamepadctl for a simple reference client.

Just supports a bidirectional shell with the MG instance, like opening its standard input remotely.

Also plays nicely with letting one's normal user account interact with an instance running as a system user.

Obviously missing a lot of potential functionality, but it's a start.

closed 2017-04-05 by jgeumlek