jgeumlek/MoltenGamepad #12: moltengamepad.cpp fails when compiling ("could not convert from brace-enclosed initializer")

Decoherent, 2016-08-18:
Hello, I was trying to set up your program to work with RetroPie, and I've run into a compiler problem. I'm really not familiar with C++, so sorry if this is something trivial!

While make is running on moltengamepad.cpp, it throws a bunch of errors similar to:

> /source/moltengamepad.cpp:180:1: error: could not convert {"num_gamepads", "Number of virtual gamepads to create", "4", MG_INT} from \<brace-enclosed initializer list\> to option_info
>   };
>  ^

This continues for the rest of the lines in `const option_info general_options[]`, which you'll forgive me if I don't retype! Is there a straight-forward way to correct this?

jgeumlek, 2016-08-19:
Hmm.

It compiles fine for me, which probably means I'm making some compiler assumption in my code.

Do you know what C++ compiler you are using?

For me, it is `g++ (GCC) 6.1.1 201608021`

jgeumlek, 2016-08-19:
Looking at [this question](http://stackoverflow.com/questions/34490692/why-do-clang-and-gcc-handle-braced-initialization-of-structs-with-in-class-initi) it might be due to `option_info` having an initializer for the `locked` variable, which affects some gcc versions.

I'll try to do a quick hotfix patch that removes that, but the compiler version info would still be nice to have.

jgeumlek, 2016-08-20:
With GCC 4.9 I got compile errors that match the one you posted.

I've just pushed a quick fix to the master branch that resolved those errors. Does it compile for you now?

Decoherent, 2016-08-20:
Sorry, I'd actually gotten angry at RetroPie for a bit, but I shouldn't abandon a bug report :)

The version of GCC installed in their image is 4.9.2-10. Looks like 4.9.2 is in the current version of Raspbian (Debian jessie). `moltengamepad.cpp` compiles happily now! `key_list.cpp` still complains about some values not being declared, notably:
```
157:KEY_ROTATE_DISPLAY
462:KEY_NUMERIC_A
463:KEY_NUMERIC_B
464:KEY_NUMERIC_C
465:KEY_NUMERIC_D
```
but once those lines are removed, make completes successfully. (what a chore compiling on an original RPI is.) Getting it working took some banging (none of the wikis are quite right), but it seems to be working great! Thanks!

jgeumlek, 2016-08-20:
Thanks for reporting this and following up.

As long as GCC 4.9 doesn't impose more significant constraints I am fine with supporting this compiler version as well.

closed 2016-08-20 by jgeumlek