jgeumlek/MoltenGamepad #16: Feature Request: Driver+Fingerprint matching method for gendev

TiZ-EX1, 2016-10-09:
Hiya. I know it's been a while since I've contributed. I think I mentioned last time a bash daemon I had that starts xboxdrv on event devices and hides them. I actually have it on github [right here](https://github.com/TiZ-EX1/rejoy). I wanted to just start using your project, but there is one problem I realized that caused me to go back to rejoy and tweak it. I'm no good at C++, it scares the crap out of me, so I couldn't just write the feature myself, sorry.

My laptop goes with me to Arcade Legacy where there are many many many many different kinds of controllers. The only kinds of controllers that tend to get introduced to my laptop are 360, PS3, and PS4 controllers, alongside the odd Mayflash PS2 to PC every once in a while. The problem is, if someone plugs in their controller and it doesn't work for any reason and/or I have to take time to configure it, it's an embarrassment to me and an embarrassment to Linux.

Names to match devices are a good start, it gets around the thing where multiple similar devices have different model IDs but the sane name, and that's good. But I feel we should be able to do better than that. If a friend plugs in a fight stick that fits a well-known event code layout but it fails to work just because its name hasn't been added to the appropriate gendevice profile, it's an embarrassing inconvenience. I came up with a way to automatically accommodate 99% of the controllers that my laptop will ever see with my script, and I'd like to pitch that method here for inclusion into this program so I can eventually stop using that particular bash monstrosity.

The first and maybe easiest thing to do is match on the device's driver. Anything using the xpad driver is always going to have the same button mapping no matter what, so rather than waiting for an xpad device not called "Microsoft X-Box 360 pad" to come out and cause trouble, we could just match the xpad driver and any device using that will use the same gendevice profile. Maybe gendevices that match on driver could have section headers like \[driver:xpad\].

That fixes the easiest case, but other drivers don't have the same convention that xpad does, not even sony. The dualshock 3 and dualshock 4 have different mappings. However... they also expose different events. They have different "fingerprints". For example, the DS3's guide button maps to key 704, which doesn't exist on the DS4. To match based on fingerprint, I load a xboxdrv evdev config and see if the new joystick exposes events to match every event specified in that config. If not, move on until there is a match. So if I plug in a DS4, it first checks the DS3 config. The DS3 config calls for 704, which doesn't exist. So it moves onto DS4, and every event called for in the config does exist on that device, so it uses that config.

This can be used for hid-generic devices too--which a lot of fight sticks tend to use even when they're technically PS3 controllers--as well as those funky Mayflash adapters. I use the same strategy here. Mayflash's PS2 adapters use a different subset of event codes--a fairly exotic one at that--from PS3 fight sticks which use a more standard set of event codes. When a PS3 fight stick gets plugged in, it will first check the Mayflash config, fail that check, and move on to the PS3-on-generic config, succeed, and use that. This fingerprint strategy relies on trying the more exotic fingerprints first, which will usually fail pretty fast and will prevent false positives.

Is this something you would have interest in implementing, or at least not be opposed to having implemented? If I were to try and get over my fear of C++ to do it, I'm a bit lost in this maze of code. What files would I want to look at to figure this out?

jgeumlek, 2016-10-10:
I totally agree that having to deal with configs is an embarrassment. I want using controllers to be as slick as possible.

Thanks for the suggestions.

The driver matching can be added quite easily. There is already support for matching vendor/product ids instead of device names. Matching on the driver would just be an extra field to check. However, it isn't true that all xpad's devices have the same mapping. The wireless controllers are mapped differently than the wired ones last I checked. It is still easy enough to add that I'll put it in.

I think fingerprinting on events is a good idea. It'll be a bit harder to add as the available events are a bit harder to extract at the time of choosing a driver. The biggest hold up I have is picking the syntax for specifying we want these matches.

Your setup just reuses the already existing config of events, and that is probably better than trying to specify exactly which events are the fingerprint.

After some thought, I'm thinking something along having it in the match line as follows:

    [events="XXXXX"]

Where "XXXXX" can be one of `subset`, `superset`, or `exact`, each imposing a constraint on how the events of the device relate to the events of the driver.

Enforcing the precedence of the drivers can be done by arranging the files, such as prefixing them with numbers. (Note: I might need to sort filenames internally to guarantee this).

TiZ-EX1, 2016-10-10:
> The wireless controllers are mapped differently than the wired ones last I checked. It is still easy enough to add that I'll put it in.

Oh heck, you don't say? Well, if they expose different event codes, we can still exploit fingerprinting.

> Your setup just reuses the already existing config of events, and that is probably better than trying to specify exactly which events are the fingerprint.

I think that's personally the most economic and least-confusing--at least in terms of user-facing--way of doing it. Implementation-wise, I can see how that would be inconvenient because you might need to make multiple passes at the configs, one for the fingerprinting and one for the config that actually gets loaded.

I think it's also important that fingerprinting is done last. Personally, the precedence I would use is model > name > vendor > driver > fingerprint. But I love the idea that you could specify superset/subset in either direction as well as exact. I think it's a great idea. I'm glad that you like this idea. :)

jgeumlek, 2016-10-11:
I wrote an initial version of this fingerprinting and put it on the devel branch. It implemented it the way I mentioned above. Inside a match line, you can put `events=` followed by subset, superset, or exact.

For example, with a DS4 I have `[events="superset"]` as one of the match lines (since the gendev driver can't do anything with the multitouch events that technically are considered axis events).

It appears to be working on my end, but I only have a couple controller types over here. Let me know how it works out for you.

However, I did not implement anyway of specifically giving these fingerprint matches a lower precedence. I've opened a separate issue on how to achieve that.

TiZ-EX1, 2016-10-11:
Cool, I'll adapt my rejoy xboxdrv configs to this new format and test. A quick question though: is there any way to specify event codes--as in numbers--rather than names, just to be sure that any given config will work everywhere?

jgeumlek, 2016-10-11:
Not currently. I could add it. Specifying event codes which don't exist on older kernels won't necessarily mean they'll work, I think.

Unfortunately `key_#306` would cause problems as the `#` comments out the rest of the line.

Which do you prefer:

* `key_?306`, with the question mark denoting the sense of uncertainty for this unnamed event.
* `"key_#306"`, that is to say these numeric events would have to be in quotes.
* Something else?

TiZ-EX1, 2016-10-11:
If the event codes don't exist on older kernels, doesn't that mean support for the device just isn't there?

Anyways, is there a particular reason we can't just have `key_306`, `abs_001`, etc? I think that'd be the cleanest way, personally.

jgeumlek, 2016-10-11:
Yes, I do believe that means that device wouldn't be supported by that kernel. Hence why I am uncertain about how allowing numeric event designations would make a config work "anywhere". It seems like we'd always want MG to be compiled with an event list that matches the system it runs on. There are some naughty devices out there that output undefined codes, so this is still a worthwhile feature.

The number keys already have `key_1` through `key_9`, and those are actually off by one relative to their values. There is also `key_102nd`, which has a value of 86 and would make the parsing a bit more annoying.

TiZ-EX1, 2016-10-11:
That's a very good point. Maybe instead of the awkward ? and #, we could just use n? `key_n306`? `abs_n001`?

jgeumlek, 2016-10-11:
I strongly prefer that it be not even remotely possible for a conflict or even momentary confusion, so I'd like it to use a character outside `a-z0-9_`.

How do you feel about `key(306)`?

TiZ-EX1, 2016-10-11:
I like that one! c:

TiZ-EX1, 2016-10-15:
Is there a way to get some sort of debug output? Right now it's really quiet, and I'm kind of feeling like it can't even see the gamepads that I'm connecting. Maybe it isn't? Is it supposed to say anything when a device appears (or rather, appears and presses a button), even if it doesn't match any profiles?

jgeumlek, 2016-10-15:
There isn't any official debug output.

You should see something printed when a device is added successfully (i.e. it matched a driver). ex. `hotplug: device ds4_1 added.`

Devices which don't get claimed do not lead to any output (on my laptop, I have 15 non-gamepad input devices just from the power button, speakers, etc. Printing each would add a fair bit of noise to startup.)

Whenever I need some debug info temporarily, I just add a few `std::cout << VARIABLE << std::endl;` statements to print out stuff in a function.

Adding a real notion of debug levels would be a good idea. Can you open an issue with what sort of debug info you'd like to see?

TiZ-EX1, 2016-10-18:
Thanks to all you've done with this feature request, I feel confident that I've replicated the big safety net I had in Rejoy inside of MoltenGamepad. The only bummer is tr2 not working if I use tr2_axis_btn on digital triggers. I have some example configs here: https://github.com/TiZ-EX1/MG-Files/tree/fingerprints

jgeumlek, 2016-10-24:
This feature finally hit master. Closed.

closed 2016-10-24 by jgeumlek