jgeumlek/MoltenGamepad #17: Generic Driver Matching Priorities

jgeumlek, 2016-10-11:
With the addition of matching generic devices based on the events reported, we have what we might consider a low priority match. It is likely best to save this as a fallback in case all other matches fail.

When MoltenGamepad sees a device, it currently goes through each driver once letting them answer yes/no on whether they want to claim that device. The search stops at the first yes. There is not a way for a driver to say "yes, but only if no other driver wants it."

So the issue is two-fold:

1. We need to decide a syntax for specifying matching priority.
2. We need to change the internals of MG to support matching priorities.

The current way of enforcing priority is by the ordering of drivers in terms of when MoltenGamepad reads them. This can prove insufficient when we have two drivers that each wish to be a fallback while also superseding the other on specific matches. The current work around would be to duplicate both drivers, one for specific matching and one for fingerprinting.

I don't want to get in the mess of hardcoding heuristics on what matching criteria gets precedence, since each match line can contain different fields and different numbers of fields. It seems messy, and likely to be wrong for some use case some where down the road.

Some ideas:

* Add a `priority=X` field to match lines, with the default being the strongest priority. This allows the event-fingerprinting lines to be given weaker priorities without requiring extra work on the other match lines. It adds some overhead to the config writer, as they now need to balance priority levels across files.
* Add negative match lines, allowing us to punch holes in an earlier driver's matches to let the later one claim it. Has the advantage of not requiring the internals to change, but it puts information into our configs far away from where it is logically relevant, and would require these holes be punched in all drivers in between as well.
* Similar to negative matches, we could try to allow for distant positive match lines, so that for specific devices we can skip to a later driver that otherwise would've had lower priority.

I think that first idea is the best of those.

TiZ-EX1, 2016-10-11:
I forgot to consider the fact that multiple fields can be in a match line. I don't blame you for not wanting to get into that mess. I like the first idea best as well. It's the most straightforward way, lets the user decide, and is least likely to blow up on account of doing something unexpected.

jgeumlek, 2016-10-11:
Ok, seems like idea one is the winner. Now onto syntax.

We want the default priority to be the max priority, and this allows us to stop at the first driver that finds a max-priority match.

Since I want an unambiguous always-the-strongest priority without putting an arbitrary limit on the number of priority levels, I'm thinking the default should be `0` or `1`, with lower priorities taking higher numbers. The user can build whatever hierarchy of generic drivers they like.

The user wouldn't be able to change the priority relative to built in drivers. This is unlikely to ever be feasible, as the built in drivers aren't as flexible in their matching. It should be considered a bug if a built in driver claims a device it shouldn't.

Is it awkward to have low priorities have high numbers? Is there perhaps a better name for this field than `priority` that still gets its purpose across?

TiZ-EX1, 2016-10-11:
It's not awkward to have high-numbered low-priorities, that's been the norm in Linux for years. Important udev rules that must run first are low numbered, and udev rules that must run last are high numbered.

jgeumlek, 2016-10-14:
The devel branch now has this feature, as an `order` field on match lines. The lowest (and default) value is `1`. After some testing it appears to be working on my end.

TiZ-EX1, 2016-10-18:
I haven't strenuously tested this, but the debug log suggests it's working fine.

commit 91455a95 (2016-10-12): add gendev matching priorities (resolve #17)
closed 2016-10-24 by jgeumlek