jgeumlek/MoltenGamepad #18: Debug information

TiZ-EX1, 2016-10-17:
I'm trying to get my rejoy configurations ported over, and the program's radio silence is causing me a lot of trouble because I have no idea what's not working or even why. I know that I could use cout << thing << endl myself, but as I said in #16, I don't even know where to begin in regards to navigating this codebase. Or any non-trivial C++ codebase, really.

Anyways, the main thing I'd like to see in regards to a debug switch is as follows: any time a device is examined, all the things that could be used to match against it should be printed out. Its name. Vendor ID. Model ID. Driver. And maybe if you wanted to go this far, if a device fails a fingerprint match, the reason why should be printed out.

This is all I can think of right now.

TiZ-EX1, 2016-10-17:
Actually, I'd like to see the entire chain of gendevice configs it goes through and why it fails each check. I can get the device added by name, but no combination of driver or events is causing the device to get claimed.

jgeumlek, 2016-10-17:
First pass at adding this debug info has been pushed to the `devel` branch.

I added a `--verbose` (or `-V`) flag to turn on the first level. You'll see every device processed by MoltenGamepad, along with every device manager that is checked.

You can repeat the flag (i.e. `--verbose --verbose`) to turn on the super detailed level. This level displays information about each match line and offers reasons for why a device was accepted/rejected.

Turns out the `driver` field was bugged: I was only checking the device's reported driver, when we really need to look at the parent devices too.

I hope this helps.

TiZ-EX1, 2016-10-18:
That commit gave me everything I needed to get my gendevice configs working! Thank you so much!

closed 2016-10-18 by TiZ-EX1