jgeumlek/MoltenGamepad #19: moltengamepad.cfg ignored when used with --config-path

TiZ-EX1, 2016-10-18:
So I moved my configs and compiled binary to a nice place where all my computers would get it. Everything in this directory is world-readable, and it has to be in this case since I have a system user/group that MG runs as. The gendevices work just fine in this configuration. But moltengamepad.cfg, which is also world-readable, is silently ignored. A github search of the source code indicates that it should say that it's attempting to load the file, but I'm not getting that output.

```
TiZLappy:~$ sudo /home/common/bin/moltengamepad -c /home/common/confdata/moltengamepad
loading 
driver: wiimote driver initialized.
driver: xpad-dt driver initialized.
driver: xpad-at driver initialized.
driver: playstation4-dt driver initialized.
driver: playstation4-at driver initialized.
driver: mayflash-ps2 driver initialized.
driver: dualshock3 driver initialized.
hotplug: device ps4_dt_1 added.
^CShutting down.
stopping udev thread
```

I did a github search of the locate function that is seemingly used to find the config file, and it looks like that function just doesn't respect the --config-path option, and only checks for the profile dir and the gendev dir. But I don't know your code, so I don't know if that means anything.

jgeumlek, 2016-10-18:
Thanks. I'll look into this when I get back to my dev machine.

In the meantime, can you try setting `$XDG_CONFIG_HOME` instead when running moltengamepad? Based on what you have above, set it to `/home/common/confdata/`. (Be sure not to use the `-c` flag when testing this.)

TiZ-EX1, 2016-10-19:
I tried multiple ways of setting it, even this inane way:

```
TiZLappy:system$ cat /tmp/molten
#!/bin/bash
export XDG_CONFIG_HOME=/home/common/confdata/
echo $XDG_CONFIG_HOME
exec /home/common/bin/moltengamepad "$@"
TiZLappy:system$ sudo /tmp/molten
/home/common/confdata/
loading 
driver: wiimote driver initialized.
^CShutting down.
stopping udev thread
```

No good. :(

TiZ-EX1, 2016-10-19:
Rather, looks like the variable is just straight up not working:

```
TiZLappy:system$ XDG_CONFIG_HOME=/home/common/confdata/ moltengamepad
loading 
driver: wiimote driver initialized.
^CShutting down.
```

jgeumlek, 2016-10-19:
Thanks. I see that the gendev drivers stopped working, and so something is definitely up. I'll be sure to fix this. It's worked for me, and I consider this an important feature.

I assume this still doesn't work even if you add the slash to the end of your original command.

As a final workaround until the fix, how about the nuclear option of a symlink from either of the two default locations?

TiZ-EX1, 2016-10-19:
Linking /etc/xdg/moltengamepad to /home/common/confdata/moltengamepad allows the system user to read the configs. This will work for now, thanks.

jgeumlek, 2016-10-21:
Okay, I think this is resolved. There were two bugs:

1) The `-c` flag did not add a path separator, so it was trying to open `/home/common/confdata/moltengamepadmoltengamepad.cfg` in your first command. Now we always add the path separator.

2) Instead of setting a string to the value of `XDG_CONFIG_HOME`, I set a string to itself! So yes, the variable was essentially being thrown away. `XDG_CONFIG_DIRS` was working fine, however.

Some extra file locating debug lines were added to the `-VV` mode just in case oddities arise in the future.

TiZ-EX1, 2016-10-21:
Both cases are now working perfectly fine. Thank you! :D

commit 2164f80d (2016-10-21): fix reading XDG_CONFIG_HOME (resolve #19)
closed 2016-10-21 by TiZ-EX1