jgeumlek/MoltenGamepad #23: Custom profile mapping not being applied

iRamos99, 2016-11-30:
I'm running MoltenGamepas as root, I know it is not the recommended but I didn't have an option with my current setup. I want to have the same mapping on the wiimote buttons with/without nunchuck connected. That way, I don't have to change the wiimote orientation when navigating the Media Center application with/without nunchuck. Everything works like a charm (other than the custom profile). I can create custom mappings on my other applications (Kodi, EmulationStation, and RetroArch). But when I disconnect the nunchuck the mappings are different because MoltedGamepad is sending different command to the buttons for each mode. 

First, I saved the default wiimote profile, cleaned it up, and made my edits. The profile file is in 'root/.config/moltengamepad/profiles/<profilename>'. Then, I created a configuration file in 'root/.config/moltengamepad/moltengamepad.cfg'. with the command `load profiles from "profilename"`. I double checked and triple checked the profile syntax and looked for typos. After trying a few times and getting no results, I tested the configuration file by also adding `mimin_xpad=true` and this command worked.

iRamos99, 2016-11-30:
The profile file that I created had this format;:
`[wiimote]
wiimote.wm_a = select`

It worked by changing it to:
`[wiimote]
wm_a = select`

closed 2016-11-30 by iRamos99

jgeumlek, 2016-11-30:
Glad to see you got it working, but I want to see if I can prevent a similar situation from arising again.

Would you mind providing some extra details? I'm having trouble reproducing what went wrong.

* If I understand correctly, the `moltengamepad.cfg` was always working, and you were seeing MG attempt to load your profile file at start-up. Yes?
* Despite loading said profile, the mappings weren't working. Ultimately this was fixed by editing the profile file. What output if any was seen when the non-working profile was being loaded?
* Would you mind posting an excerpt of the original non-working profile file?

The following should be valid, both with and without the "wiimote." on the second line:

    [wiimote]
    wiimote.wm_a = select

or

    [wiimote]
    wm_a = select

iRamos99, 2016-12-02:
Thanks for replying. I was mistaken on what made it work. I ran some test after viewing your reply to make sure that I was correct. Well, I wasn't. The problem was that the configuration files that I was trying to get working lived in
`
root/.config/moltengamepad/profiles/mwprofile
root/.config/moltengamepad/moltengamepad.cfg
`
But I also had configuration files in
`
/etc/xdg/moltengamepad/profiles/mwprofile
/etc/xdg/moltengamepad/moltengamepad.cfg
`
I was editing the files in the `root/.config/...` directory, but no changes seem to take. Then, when I removed the `wiimote.` prefix from the input mapping, I also copied the filed to the `etc/xdg/...` directory. That time it worked and I assumed that it was because of removing the `wiimote.` prefix. When in fact it turns out that the files that are being loaded by MG are the files in `/etc/xdg/...` .

Anyway, here is my configuration:
User: root

Code to load on boot:`
Directory: /lib/systemd/system/moltengamepad.service
``
[Unit]
Description=MoltenGamepad
After=network.target
After=bluetooth.target

[Service]
Type=forking
PIDFile=/var/run/moltengamepad.pid
Environment="XDG_CONFIG_HOME=/etc/moltengamepad"
ExecStart=/usr/sbin/moltengamepad --daemon --pidfile /var/run/moltengamepad.pid
ExecStop=/usr/bin/kill $MAINPID

[Install]
WantedBy=default.target
```

Configuration files directories:
```
/etc/xdg/moltengamepad/profiles/wmprofile_old
/etc/xdg/moltengamepad/profiles/mwprofile
/etc/xdg/moltengamepad/moltengamepad.cfg
```

Code in moltengamepad.cfg
```
#useful settings
#mimic_xpad = true
#make_fifo = true
    
#load preferred default mappings
#load profiles from "wmprofile_old"
load profiles from "wmprofile"

```
Code in previous profile version: (confirmed working)
```
[wiimote]
wiimote.cc_a = primary
wiimote.cc_b = secondary
wiimote.cc_x = third
wiimote.cc_y = fourth
wiimote.cc_plus = start
wiimote.cc_minus = select
wiimote.cc_home = mode
wiimote.cc_thumbl = thumbl
wiimote.cc_l = tl
wiimote.cc_zl = tl2
wiimote.cc_thumbr = thumbr
wiimote.cc_r = tr
wiimote.cc_zr = tr2
wiimote.cc_up = up
wiimote.cc_down = down
wiimote.cc_left = left
wiimote.cc_right = right
wiimote.cc_left_x = +left_x
wiimote.cc_left_y = +left_y
wiimote.cc_right_x = +right_x
wiimote.cc_right_y = +right_y
wiimote.wm_a = primary
wiimote.wm_b = secondary
wiimote.wm_1 = fourth
wiimote.wm_2 = third
wiimote.wm_plus = start
wiimote.wm_minus = select
wiimote.wm_home = mode
wiimote.wm_up = up
wiimote.wm_down = down
wiimote.wm_left = left
wiimote.wm_right = right
wiimote.wm_ir_y = nothing
wiimote.wm_ir_x = nothing
wiimote.wm_accel_x = nothing
wiimote.wm_accel_y = nothing
wiimote.wm_accel_z = nothing
wiimote.nk_a = primary
wiimote.nk_b = secondary
wiimote.nk_1 = fourth
wiimote.nk_2 = third
wiimote.nk_c = tl
wiimote.nk_z = tr
wiimote.nk_plus = start
wiimote.nk_minus = select
wiimote.nk_home = mode
wiimote.nk_up = up
wiimote.nk_down = down
wiimote.nk_left = left
wiimote.nk_right = right
wiimote.nk_stick_y = +left_y
wiimote.nk_stick_x = +left_x
wiimote.nk_ir_x = nothing
wiimote.nk_ir_y = nothing
wiimote.nk_accel_x = nothing
wiimote.nk_accel_y = nothing
wiimote.nk_accel_z = nothing
wiimote.nk_wm_accel_x = nothing
wiimote.nk_wm_accel_y = nothing
wiimote.nk_wm_accel_z = nothing
wiimote.bal_x = nothing
wiimote.bal_y = nothing
?grab_permissions = false
	#  Grab device via blocking all read permissions
?grab_exclusive = true
	#  Grab device events via ioctl EVIOCGRAB
?nk_ir_active = false
	#  Enable IR data when nunchuk is present
?wm_ir_active = false
	#  Enable IR data when no extension is present
?nk_accel_active = false
	#  Enable accelerometers when nunchuk is present
?wm_accel_active = false
	#  Enable accelerometers when no extension is present
```

Code in current profile version: (confirmed working also)
```
[wiimote]
cc_a = primary
cc_b = secondary
cc_x = third
cc_y = fourth
cc_plus = start
cc_minus = select
cc_home = mode
cc_thumbl = thumbl
cc_l = tl
cc_zl = tl2
cc_thumbr = thumbr
cc_r = tr
cc_zr = tr2
cc_up = up
cc_down = down
cc_left = left
cc_right = right
cc_left_x = +left_x
cc_left_y = +left_y
cc_right_x = +right_x
cc_right_y = +right_y
wm_a = primary
wm_b = secondary
wm_1 = fourth
wm_2 = third
wm_plus = start
wm_minus = select
wm_home = mode
wm_up = up
wm_down = down
wm_left = left
wm_right = right
wm_ir_y = nothing
wm_ir_x = nothing
wm_accel_x = nothing
wm_accel_y = nothing
wm_accel_z = nothing
nk_a = primary
nk_b = secondary
nk_1 = fourth
nk_2 = third
nk_c = tl
nk_z = tr
nk_plus = start
nk_minus = select
nk_home = mode
nk_up = up
nk_down = down
nk_left = left
nk_right = right
nk_stick_y = +left_y
nk_stick_x = +left_x
nk_ir_x = nothing
nk_ir_y = nothing
nk_accel_x = nothing
nk_accel_y = nothing
nk_accel_z = nothing
nk_wm_accel_x = nothing
nk_wm_accel_y = nothing
nk_wm_accel_z = nothing
bal_x = nothing
bal_y = nothing
?grab_permissions = false
#  Grab device via blocking all read permissions
?grab_exclusive = true
#  Grab device events via ioctl EVIOCGRAB
?nk_ir_active = false
#  Enable IR data when nunchuk is present
?wm_ir_active = false
#  Enable IR data when no extension is present
?nk_accel_active = false
#  Enable accelerometers when nunchuk is present
?wm_accel_active = false
#  Enable accelerometers when no extension is present

```

jgeumlek, 2016-12-02:
Thanks a ton for providing all that info. Yes, both of those profile files seem fine.

Though I think I can see where things can go awry with that service file.

1) You set `$XDG_CONFIG_HOME` to `/etc/moltengamepad`. When this is set, MG ignores `$HOME` and just uses that directory (so it will never look in `/root/`). However, with that value, MG will expect files in `/etc/moltengamepad/moltengamepad/` which also probably isn't what you want.

2) Even if `$XDG_CONFIG_HOME` wasn't set, I don't think I trust service files to always set `$HOME` of root to `/root/`. I think it might be unset.

However, regardless of the above, MG was respecting `$XDG_CONFIG_DIRS` which defaults to a fallback of `/etc/xdg/` (and thus files in `/etc/xdg/moltengamepad/`).

For MG-as-root, I would recommend just using `/etc/xdg/moltengamepad` and ignoring `/root/`.

jgeumlek, 2016-12-02:
Hmm, as a follow-up:

When running as root or a system user, I don't really know where MG should save profiles when instructed by the user.

 Doing a home directory for root seems wrong, but so does writing to `/etc/`.

Maybe MG should require absolute paths in that case.