jgeumlek/MoltenGamepad #24: Wiimote nunchuck way too sensitive

qwertychouskie, 2016-12-14:
Pushing the joystick on the nunchuck up about a third of the way tells the software it is pushed up all the way.

jgeumlek, 2016-12-15:
Good catch. It appears I copied the stick ranges from the classic controller, but they don't match the nunchuk stick.

I'll put out a hotfix on the master branch for this soon after I test out my various nunchuks, but in the meantime, you can change this in `source/plugin/wiimote/wii_events.cpp`, line 230:

    //the current code:
    #define NUNCHUK_STICK_SCALE ABS_RANGE/24

The 24 should be changed to whatever value should max-out the stick. Based off of a quick test with one nunchuk, 85 or 90 appear to reasonable values.

    //the new code:
    #define NUNCHUK_STICK_SCALE ABS_RANGE/85

jgeumlek, 2016-12-20:
Should be fixed in v0.3.3

Some day, it would be nice to have a centralized way of handling these axis scalings and calibration in MG.

closed 2016-12-20 by jgeumlek