jgeumlek/MoltenGamepad #25: Can you make it that I can select per process what input is received

jappeace, 2016-12-21:
This would basically add split screen support to any game that can setup local networking (and that can run multiple process instances of course, I think some block this).
so it would be something like:

/dev/controller1 -> pid10
/dev/controller2 -> pid20

I want to do this myself but I honestly don't know that much about the lower levels.

jgeumlek, 2016-12-21:
I'm not sure I follow what you are asking.

I think you are asking for a way to control which devices a process opens, such that say Program A might be  only reading controller 1, while Program B only reads controller 2.

If so, this isn't under MoltenGamepad's control. MG creates virtual devices, but it can't control who opens them. The file permissions on the `/dev/input/event#` nodes control that.

Unfortunately, I don't see much of a way to include this in MG, though I think I can see why it would be useful.

If you want to achieve that, I think you could by fiddling with permissions, without MG. 

Idea one: Run Prog A with only controller one readable, change permissions to make only controller two readable, run Prog B. (Prog A will still have controller one open and readable even if permissions change).

Idea two: Run Prog A and Prog B as two separate users. Set up controller permissions so that each can only open and read the appropriate controller.

closed 2016-12-21 by jgeumlek

jappeace, 2016-12-22:
Ok thanks for your ideas, I'll try to do this and probably embed it into a script.