jgeumlek/MoltenGamepad #27: Mapping d-pad to LS

TiZ-EX1, 2017-01-22:
Hi there. So, I have a couple old games that only pay attention to the left stick, as well as games where it would benefit to be able to use either method. The impression I got from the documentation is that I should be able to load a profile to do this, something like this:

```
gamepad.left = -left_x
gamepad.right = +left_x
gamepad.up = -left_y
gamepad.down = +left_y
```

If I echo `load profiles from dpad_to_ls`, the log indicates it does do that, but testing the controller in antimicro shows the dpad is still sending dpad inputs and not left stick inputs. Am I doing this wrong?

In addition, I'm not clear on how I would revert these changes when I'm done playing a game that would use this.

Thanks for everything, I've been trucking along with zero trouble since the last time I came here. :D

jgeumlek, 2017-01-23:
That looks good. Does the remapping work for other output events? Or is just the stick mapping giving issues?

My guess is that you are using a controller that doesn't have four digital dpad events (left,right,up,down), but instead has two axes ("leftright" and "updown"). You should set all six events in order to handle both devices with four dpad events and devices with two dpad axes.

    gamepad.left = -left_x
    gamepad.right = +left_x
    gamepad.up = -left_y
    gamepad.down = +left_y
    gamepad.leftright = left_x
    gamepad.updown = left_y

I felt that trying to automagic axes into separate digital events would be a bit too much.

To reset, just set them back to their original mappings:
    gamepad.leftright = (left,right)
    gamepad.left = left
    # ... etc.

When in doubt, use `print profile <device>` to see the profile in use, after you've done the mappings above. Based on aliasing, it might differ from the "gamepad" profile.

TiZ-EX1, 2017-01-23:
Thanks for replying.

I haven't tried remapping other stuff, just the d-pad because it's the only practical application for my use cases at the moment.

Adding in the leftright and updown entries makes it work... kinda. The inputs don't stay active when I hold a dpad direction down. In antimicro I see the direction light up for a few milliseconds then immediately go off.

Reverting the mapping change works fine, though.

Also, tangentially related: I remembered that I have to chmod the fifo in order to send commands to it. I'd like the fifo to be world-writeable so that normal users can configure it at runtime; can we make a config option for that? I can make a separate issue for that if you'd like.

jgeumlek, 2017-01-23:
I think the not staying active issue might be it being clobbered. A thumb stick might be having some jitter near zero, and those new near zero values overwrite the events the dpad.

The current devel branch reworked how thumb sticks are processed, including filtering out spurious near zero events in the deadzone. The new filtering should be present when the profile looks like

    gamepad.left_x = nothing
    gamepad.left_y = nothing
    gamepad.(left_x,left_y) = stick(left_x,left_y)

which is the new default. Note that the two raw axes are unmapped, and instead there is a translator that works on them jointly.

As for FIFO permissions, the devel branch also changed the umask so that the created FIFO has group-write permissions. I think that is good enough for the usability/complexity trade-off, and for now the user can just directly add more permissions after creation.

If there is a strong reason for adding config options for the permissions, I'd be willing to reconsider that decision.

TiZ-EX1, 2017-01-24:
The devel branch fixes all my problems once again, haha! After building devel, my dpad-to-left-stick profile works just fine with some cursory testing. I also think group write for the FIFO is an absolutely sensible default. I've added my normal user to the gamepad group and that works just fine too, without letting the normal user see the original gamepads.

Thanks again. I imagine the profile(s) I've made for this use case will be useful to multiple users; want me to make a pull request against MG-Files?

closed 2017-01-24 by TiZ-EX1

jgeumlek, 2017-01-25:
Glad to hear it. The devel branch should be reaching master "soon". There's still some kinks in the syntax I'd like to work out about the whole translating-events-jointly thing where the left side needs multiple input events..

Yes, please submit that PR.