jgeumlek/MoltenGamepad #29: Improve Wii Remote motion controls

jgeumlek, 2017-04-05:
Extending issues #21 and #28. Wii remotes have a lot of sensors, and it is a shame to not use them fully. Currently MG does very basic processing to emit these events.

This issue is likely to be open for a long time, as there will always be room for improvement.

Asking for some help, be it coding or testing to see how "fun" the mappings are. My limited time on this project is likely to go towards improving the framework/enabling new use cases rather than fine-tuning input translations. Play testing takes a lot more time!

You don't need to know anything low-level to help code this. MG will handle reading in all the sensor data, and sending the output values to the virtual devices. 

If you were given the sensor values, can you compute sensible output values?

All of them require some sort of smoothing: the raw data is far too jittery to use in most games. What smoothing is best?

For the motion-plus gyros, we have a suggested way to generate relative mouse movements. See #28 for one implementation, and the "gyro mouse" project of this repo for the progress towards merging this into MG.

For tilt support, does it make sense to emit the accel data, or do some trig to emit an estimated angle? Changing the semantics to an angle also gives us more freedom to handle the case where you turn so far the controller goes past vertical or upside down...

gustavolinux, 2017-04-09:
Tell me what values in the code need to be changed to start fine-tuning the wimote relative input. I will be glad to help.

jgeumlek, 2017-04-12:
On the devel branch, I added my implementation of the logic I saw in Ryochan7's gist.

Wiimote calibration occurs near the bottom of wii_events.cpp. The rest of the motion plus processing code is in `core/event_translators/group/wiigyromouse.cpp'. However, most of wiigyromouse's constants can be chosen during run time.

This involved adding the motion+ gyro events to the wiimote input sources, along with device options to enable them (much like how IR/accelerometers are currently handled).

This new translator does not automatically redirect the events to the keyboard slot, so be sure to move the wiimote to the keyboard slot.

If you wish to test it without a nunchuk:

    wiimote.?wm_gyro_active = true
    wiimote.?wm_accel_active = true
    wiimote.(wm_accels,wm_gyros) = wiigyromouse
    #Optional: click events
    # wiimote.wm_a = btn_left
    # wiimote.wm_b = btn_right
    move all to keyboard

With a nunchuk, the first few lines should look like

    wiimote.?nk_gyro_active = true
    wiimote.?nk_accel_active = true
    wiimote.(nk_wm_accels,nk_wm_gyros) = wiigyromouse
    # wiimote.nk_a = btn_left

You can optionally tack on some button events after the gyros. The first button will be a ratchet (no cursor movements when held, imaging lifting a mouse from a mouse pad). Any further buttons are dampening buttons: movements are dampened temporarily after seeing a button press.

    wiimote.(wm_accels,wm_gyros,wm_down,wm_a,wm_b) = wiigyromouse
    #wm_down acts as a ratchet button
    #wm_a and wm_b will cause dampening

You might want to try negating the ratchet button: movements will only be generated when the ratchet is held.

    #use negated wm_a as the ratchet: move mouse only when wm_a is held.
    wiimote.(wm_accels, wm_gyros, wm_a-) = wiigyromouse

There are a lot of options exposed by `wiigyromouse`. The full expression:

    wiigyromouse(float smooth_factor=.75, int buffer_size=4, float xscale=1, float yscale=1, float min_offset=0.0616205, float max_offset=60, float deadzone = .01, bool use_accels=true, float dampen_factor = 0, int dampen_period = 1)

Smooth_factor affects the averaging: As it goes to 0 it puts all the weight on the newest value, as it goes to 1 it puts more weight on the old values.

buffer_size: the amount of past values that are averaged together. Choosing buffer_size == 1 is special: a running average is computed, which acts like an infinite buffer size, but it does mean it accumulates errors.

xscale,yscale: allows the horz/vertical movements to be scaled independently

deadzone: a value between 0 and 1. When the new values are within the deadzone of their output range, we pretend we saw just 0s.

min_offset/max_offset: The final movements outputted should range smoothly between these extremes (i.e. when we barely pass the deadzone threshhold, we output min_offset)

use_accels: If true, use the most recent accel data to "unrotate" the observed movements. Try to see if the controller is sideways/upside down and compensate.

dampen_factor: A scaling factor on all observed movements during a dampening period

dampen_period: The length in milliseconds after a dampening button is pressed that we should dampen movements.

chanilino, 2017-06-13:
Hi! 

I have implemented a new [pull request](https://github.com/jgeumlek/MoltenGamepad/pull/39) to calibrate online of analog controls. Perhaps this could help to calibrate the controls.