jgeumlek/MoltenGamepad #30: Add LED support

jgeumlek, 2017-04-05:
The WiimoteGlue project that came before MG supported changing the wiimote LEDs to match their assigned output slot. It's time for MG to regain that.

This will require specialized drivers for each input source that supports this, as there really isn't a generic interface for setting controller LEDs.

In terms of API, pass each device a number to display. Leave it up to the driver to decide what LED pattern/color is appropriate.

In the future, possibly add a way to query a device for either an LED pattern or color, so that clients can display it...

afiction, 2017-10-07:
Looking forward to this. I hope it's still in the works.

jgeumlek, 2017-10-16:
Development has stalled a tad due to grad school, but this is still on the list of features for the next update.

This feature is waiting on #31, since the implementation of dynamic output slots will influence the design of how MG should handle slot LEDs.