jgeumlek/MoltenGamepad #37: Debian builds, warnings being treated as errors.

mdeguzis, 2017-05-14:
Love it or hate it, Debian dpkg-buildpackage/debuid has some rulesets when building, some warnings are treated as errors for instance.

```
        make -j1
make[1]: Entering directory '/build/molten-gamepad-1.1.0+git+bsos'
compiling source/core/commandline.cpp...
mv -f .d/source/core/commandline.Td .d/source/core/commandline.d
compiling source/core/config_parse.cpp...
mv -f .d/source/core/config_parse.Td .d/source/core/config_parse.d
compiling source/core/main.cpp...
mv -f .d/source/core/main.Td .d/source/core/main.d
compiling source/core/moltengamepad.cpp...
source/core/moltengamepad.cpp: In function 'int clear_fifo(const char*, bool)':
source/core/moltengamepad.cpp:217:51: warning: ignoring return value of 'ssize_t write(int, const void*, size_t)', declared with attribute warn_unused_result [-Wunused-result]
       write(fifo,"\nexit_process_leave_fifo\n",25);
                                                   ^
source/core/moltengamepad.cpp:222:31: warning: ignoring return value of 'ssize_t write(int, const void*, size_t)', declared with attribute warn_unused_result [-Wunused-result]
       write(fifo,"\nquit\n",6);
                               ^
mv -f .d/source/core/moltengamepad.Td .d/source/core/moltengamepad.d
compiling source/core/options.cpp...
mv -f .d/source/core/options.Td .d/source/core/options.d
compiling source/core/output_slot.cpp...
source/core/output_slot.cpp: In member function 'virtual void virtual_keyboard::take_event(input_event)':
source/core/output_slot.cpp:148:37: warning: ignoring return value of 'ssize_t write(int, const void*, size_t)', declared with attribute warn_unused_result [-Wunused-result]
     write(mouse_fd, &in, sizeof(in));
                                     ^
source/core/output_slot.cpp:151:36: warning: ignoring return value of 'ssize_t write(int, const void*, size_t)', declared with attribute warn_unused_result [-Wunused-result]
   write(uinput_fd, &in, sizeof(in));
mv -f .d/source/core/output_slot.Td .d/source/core/output_slot.d
compiling source/core/parser.cpp...
mv -f .d/source/core/parser.Td .d/source/core/parser.d
compiling source/core/plugin_loader.cpp...
mv -f .d/source/core/plugin_loader.Td .d/source/core/plugin_loader.d
compiling source/core/profile.cpp...
mv -f .d/source/core/profile.Td .d/source/core/profile.d
compiling source/core/slot_manager.cpp...
mv -f .d/source/core/slot_manager.Td .d/source/core/slot_manager.d
compiling source/core/udev.cpp...
source/core/udev.cpp: In destructor 'udev_handler::~udev_handler()':
source/core/udev.cpp:90:44: warning: ignoring return value of 'ssize_t write(int, const void*, size_t)', declared with attribute warn_unused_result [-Wunused-result]
     write(pipe_fd, &signal, sizeof(signal));
                                            ^
source/core/udev.cpp: In member function 'int udev_handler::read_monitor()':
source/core/udev.cpp:167:14: warning: ignoring return value of 'int pipe(int*)', declared with attribute warn_unused_result [-Wunused-result]
   pipe(pipes);
              ^
mv -f .d/source/core/udev.Td .d/source/core/udev.d
compiling source/core/uinput.cpp...
source/core/uinput.cpp: In member function 'int uinput::make_gamepad(const uinput_ids&, bool, bool, bool)':
source/core/uinput.cpp:71:71: error: format not a string literal and no format arguments [-Werror=format-security]
   snprintf(uidev.name, UINPUT_MAX_NAME_SIZE, ids.device_string.c_str());
                                                                
source/core/uinput.cpp:71:71: error: format not a string literal and no format arguments [-Werror=format-security]
source/core/uinput.cpp: In member function 'int uinput::make_keyboard(const uinput_ids&)':
source/core/uinput.cpp:150:71: error: format not a string literal and no format arguments [-Werror=format-security]
   snprintf(uidev.name, UINPUT_MAX_NAME_SIZE, ids.device_string.c_str());
                                                                       ^
source/core/uinput.cpp:150:71: error: format not a string literal and no format arguments [-Werror=format-security]
source/core/uinput.cpp: In member function 'int uinput::make_mouse(const uinput_ids&)':
source/core/uinput.cpp:206:71: error: format not a string literal and no format arguments [-Werror=format-security]
   snprintf(uidev.name, UINPUT_MAX_NAME_SIZE, ids.device_string.c_str());
                                                                       ^
source/core/uinput.cpp:206:71: error: format not a string literal and no format arguments [-Werror=format-security]
source/core/uinput.cpp: In member function 'int uinput::make_gamepad(const uinput_ids&, bool, bool, bool)':
source/core/uinput.cpp:129:35: warning: ignoring return value of 'ssize_t write(int, const void*, size_t)', declared with attribute warn_unused_result [-Wunused-result]
   write(fd, &uidev, sizeof(uidev));
                                   ^
source/core/uinput.cpp: In member function 'int uinput::make_keyboard(const uinput_ids&)':
source/core/uinput.cpp:187:35: warning: ignoring return value of 'ssize_t write(int, const void*, size_t)', declared with attribute warn_unused_result [-Wunused-result]
   write(fd, &uidev, sizeof(uidev));
                                   ^
source/core/uinput.cpp: In member function 'int uinput::make_mouse(const uinput_ids&)':
source/core/uinput.cpp:232:35: warning: ignoring return value of 'ssize_t write(int, const void*, size_t)', declared with attribute warn_unused_result [-Wunused-result]
   write(fd, &uidev, sizeof(uidev));
                                   ^
cc1plus: some warnings being treated as errors
```

jgeumlek, 2017-05-14:
Those rules seem reasonable, those sections should have some error checking and I should have gotten rid of snprintf when I stopped numbering the uidev names. Fixing those warnings won't be a huge task.

For most of those calls, there isn't a whole lot to be done if they fail though.

mdeguzis, 2017-05-14:
As long as the warnings treated as errors are fixed, I'm fine with that. Thank you for your input.

jgeumlek, 2017-05-14:
Thanks for bringing this to my attention.