jgeumlek/MoltenGamepad #42: Sony DS4 Controller Left Axis controlling mouse and duplicate inputs

RyanFDev, 2017-10-04:
When I use the left analog stick it controls the mouse. It also sends the correct gamepad inputs but moving the mouse is annoying and not sure if intended.

I am also receiving duplicate inputs as if my original controller is not being locked properly.

Running Mint 18.2 with kernel 4.13.4

I have the proper sony generic config for the new kernel gamepad updates so that doesn't appear to be the problem.

Any help would be greatly appreciated!

amagnoni, 2017-11-07:
i have a dualshock 3, it works great in manjaro simulating a xbox 360 controller, but the new kernel 4.13 has changes in the driver, so now its showing two devices (one for the tradicional pad and the other only for the motion sensors), so it stopped working...
[image]

maybe its the case, try booting with a older kernel!
i have a 4.11 and 4.9, both works fine!

jgeumlek, 2017-11-17:
I don't have any DS3s, so I can't personally test how the new `sony` kernel module changes affect them. I hoped `sony-standard.cfg` would capture any needed changes.

I guess it might be time to be much more aggressive in hiding related devices in order to hide the motion sensors as well. However, it might be nice to not hide the DS4 touchpad...