jgeumlek/MoltenGamepad #43: Ubuntu 16.04, changing permissions on js* and event* fails

mwsduelle, 2017-10-16:
Installed as systemuser. Everything works with "sudo moltengamepad". Udev rules appear to be non-functional (had to manually chmod uinput; previously was getting "open uinput: permission denied").

```
moltengamepad
No moltengamepad.cfg found.
driver: playstation4 driver initialized.
device hiding: changing permissions of /dev/input/event11 failed.
device hiding: changing permissions of /dev/input/js0 failed.
plug: ps4_1 added
stdin: ready to read commands from standard input. Try "help" for more info.
```

gustavolinux, 2017-10-16:
try: 
`sudo chown -R youruser /dev/input/`

jgeumlek, 2017-10-16:
MG needs to own those file nodes in order to change the permissions, so it does appear your udev rules aren't working.

I don't think manually chowning the dev nodes will work, since MG only tries once to change the permissions when it first opens the device upon connection.

I don't know much about how chroots interact with the linux input system and udev rules, so this is a bit out of my depth. [This page talks about crouton and udev.](https://github.com/dnschneid/crouton/wiki/udev%3A-manage-inserted-devices) You might be able to craft a working set of udev rules if you put a little time into it.

If you can't get udev rules working on your setup, you can just ignore these errors, and not use MG's device hiding feature.

Not having the auto-hiding isn't the end of the world, since you can manually do it before launching a game.