jgeumlek/MoltenGamepad #44: Creating a Wii U Pro Profile

ROMSY-SAN, 2017-10-31:
Hey,
so i feel a little bit stupid, cause i trying to create a profile for a wii U Pro Controller. (which i would love to upload here so nobody got the same trouble as me)
But it seems i dont understand anything here...

Well first i installed MoltenGamepad and after starting it with the command ./moltengamepad --mimic-xpad i realized that the games i tried recognize the gamepad as Mouse (horray for progress!).

But back to the Problem: after pressing a button i get the following message:
`No moltengamepad.cfg found.
driver: wiimote driver initialized.
plug: wm1 added
device hiding: changing permissions of /dev/input/event25 failed.
device hiding: changing permissions of /dev/input/js4 failed.
stdin: ready to read commands from standard input. Try "help" for more info.
slot: wm1 assigned to slot virtpad1`

then i typed "print devices wm1"
and got the following output:
`wm1
 "Wii U Pro Controller"
 type: gamepad
 uniq: "------" (i removed it)
 phys: "-----"  (i removed it too)
 events:
  btn   cc_a:   Classic Controller A button
  btn   cc_b:   Classic Controller B button
  btn   cc_x:   Classic Controller X button
  btn   cc_y:   Classic Controller Y button
  btn   cc_plus:    Classic Controller + button
  btn   cc_minus:   Classic Controller - button
  btn   cc_home:    Classic Controller Home button
  btn   cc_left:    Classic Controller Dpad left
  btn   cc_right:   Classic Controller Dpad right
  btn   cc_up:  Classic Controller Dpad up
  btn   cc_down:    Classic Controller Dpad down
  btn   cc_l:   Classic Controller L button
  btn   cc_zl:  Classic Controller ZL button
  btn   cc_r:   Classic Controller R button
  btn   cc_zr:  Classic Controller ZR button
  btn   cc_thumbl:  Classic Controller Left Stick Click (Wii U Pro only)
  btn   cc_thumbr:  Classic Controller Right Stick Click  (Wii U Pro only)
  axis  cc_left_x:  Classic Controller Left Stick X
  axis  cc_left_y:  Classic Controller Left Stick Y
  axis  cc_right_x: Classic Controller Right Stick X
  axis  cc_right_y: Classic Controller Right Stick Y
 options:
?grab_exclusive = true
    Grab device events via ioctl EVIOCGRAB
?grab_permissions = true
    Grab device via blocking all read permissions`

well... now is the problem: i don't know what to do now? The main goal is to get the wiiU Pro pad recognized as xbox360 controller (or something simliar) but i don't know what to do next.

ROMSY-SAN, 2017-11-16:
ok i've think i've finally managed everything but now i don't really understand to add the device to this config.
after putting in: print drivers wiiu-pro
i've got
```
(wiiu-pro)
  no devices
```
as output and cant find a option to set my controller from 
```
(wiimote)
wm1:	Wii U Pro Controller
```
to
```
(wiiu-pro)
  no devices
```

jgeumlek, 2017-11-17:
Sorry for the delay.

The wiimote driver has built in support for the Wii U Pro, and it appears to be working as intended. It is being assigned to `virtpad1`, the first virtual output device. With `--mimic-xpad`, this should be functioning as a 360 controller, and the default mapping should be reasonable.

My Wii U Pro controller is actually the one I use most often with MG.

----
You do have some permissions issues, so the original Wii U Pro device node (`/dev/input/event25`) isn't being hidden. This means your games are going to see both the original node and the new fake 360 controller node.

Let me know what udev rules you installed and I can try to figure this out.

----
Your mouse issues are likely your desktop environment's fault. For some reason many popular distros default to treating gamepads/joysticks as a mouse input device.

----

If you wish to use your `wiiu-pro` config rather than the support in the `wiimote` driver, you'll have to remove the wiimote plugin. This can be done by changing the `Makefile` before compiling.

MG currently does not have support for rebinding devices to different MG drivers.

ROMSY-SAN, 2017-11-18:
Hey No Problem at least i could use the days to learn something new, also i hopped from Antergos to pure Arch Linux.
So if i understood you correctly the main problem is my arch linux itself.

---
For testing i tried for testing purpose moltengamepad as root:
`$ sudo ./moltengamepad --mimic-xpad`
`warning: Running MoltenGamepad as root is not recommended. Instead, please run as a non-superuser that has been granted the appropriate permissions.`
`No moltengamepad.cfg found.`
`driver: wiimote driver initialized.`
`plug: wm1 added`
`stdin: ready to read commands from standard input. Try "help" for more info.`
`slot: wm1 assigned to slot virtpad1`

so hiding is possible as root.

---

i am still try to figure out why my arch still thinking my Controler is a mouse even with --mimic-xpad 
and i didn't changed anything from your main configuration, i am more likely a "mainstrem Linux user" than somebody who really understands the most stuff (makepkg is still a little bit a hassle for me)
if i don't find any solutions why Arch is thinking my Controller is a mouse i try to use my `wiiu-pro` cfg.

jgeumlek, 2017-11-18:
For the mouse issue, see [the Arch Wiki](https://wiki.archlinux.org/index.php/Gamepad#Disable_joystick_from_controlling_mouse). Create the needed file and reboot.

----
For the permissions, for auto-hiding to work, the device nodes `/dev/input/eventX` need to have MG's user as the owner. Running as root of course gives the needed permissions also.

udev rules can be created in order to automatically assign ownership of the device nodes when they are created. The `installation` directory in this repo has a couple suggested udev rules files (along with scripts to install them).

ROMSY-SAN, 2017-11-19:
> For the permissions, for auto-hiding to work, the device nodes /dev/input/eventX need to have MG's user as the owner. Running as root of course gives the needed permissions also.

got now 3 configs:
`Section "InputClass"
        Identifier "joystick catchall"
        MatchIsJoystick "on"
        MatchDevicePath "/dev/input/event*"
        Driver "joystick"
        Option "StartKeysEnabled" "False"       #Disable mouse
        Option "StartMouseEnabled" "False"      #support
EndSection`

`Section "InputClass"
        Identifier "joystick catchall"
        MatchIsJoystick "on"
        MatchDevicePath "/dev/input/js*"
        Driver "joystick"
        Option "StartKeysEnabled" "False"       #Disable mouse
        Option "StartMouseEnabled" "False"      #support
EndSection`

and:

`Section "InputClass"
        Identifier "Nintendo Wii Remote Raw Input Blacklist"
        MatchProduct "Nintendo Wii Remote"
        MatchDevicePath "/dev/input/event*"
        Option "Ignore" "on"
EndSection`

it seems like something i should ask in the arch forum because when i connect via bluetoothctl i've got still the same issue.

ROMSY-SAN, 2017-11-19:
``

jgeumlek, 2017-11-19:
If you are still seeing it move the mouse, I don't know what is up. X11 config files like those given on the Arch Wiki are still a little bit of arcane magic to me.

However, there is still a remaining option if all else fails. Run [xinput](https://www.archlinux.org/packages/?name=xorg-xinput) to list all the current mouse/keyboard devices your desktop is seeing.

If the configs above are failing, you'll see the virtual gamepads in the list. You can then use `xinput --disable [##]`, which will tell X to ignore the mouse movements from these devices. This should work, but it'll need to be done whenever you restart MG. The config files above are supposed to do this disabling automatically.

ROMSY-SAN, 2017-11-27:
hey, first of al, i think we can close this topic, because it was all about config files and not my personal "why doesn't it work on arch" problems.. overall i asked also in the arch forums and hope for an answere.

secondly, xinput --disable doesn't seemed to help. but overall i realized, that the only thing that helped was to start FIRST moltengamepad and THEN start bluetooth and activate the controller. it still doen't work on games wich are mouse based (like 12 is better than 6) but now i can at least play all controller based games like risk of rain or guacamelee. funny thing is only that it don't really map it as a xbox controller. when i press the b button on the wii u pro pad, it doesnt realize "ok this is the a button on the xbox" it trys to tell me that is the B button on the xbox, but at least i can play more than before :)

closed 2017-11-27 by ROMSY-SAN

jgeumlek, 2017-11-27:
About that last thing:

MG's default mapping is to match by meaning, so the "B button" on the Wii U pro becomes the virtual "B button" of the xbox. This was an intentional decision.

Since the layouts are different, it isn't clear whether one might care more about the button labels (a,b,x,y) or the button positions.

If you choose to delve deeper into MG configuration, you can change the mappings, if you want the Wii U pro's button layout to match the 360 one. This is handy for games that focus on the physical layout, like Oxenfree.