jgeumlek/MoltenGamepad #45: Nunchuck thumb strick map is not working

gustavolinux, 2017-11-21:
Hi, I was trying to get my nunchuck to work with my wiimote, the C and Z buttons are working fine, but no output from my thumb stick actions..

This is my profile:
```
[wiimote]
wiimote.?nk_ir_active = true
wiimote.?nk_accel_active = true
wiimote.nk_a = key(btn_left)
wiimote.nk_b = key(btn_right)
wiimote.nk_1 = key(key_8)
wiimote.nk_2 = key(key_9)
wiimote.nk_up = key(key_o)
wiimote.nk_down = key(key_y)
wiimote.nk_left = key(key_u)
wiimote.nk_right = key(key_i)
wiimote.nk_plus = key(key_leftalt)
wiimote.nk_minus = key(key_7)
wiimote.nk_home = key(key_h)
wiimote.nk_ir_x = mouse(abs_x)
wiimote.nk_ir_y = mouse(abs_y)
wiimote.nk_c = key(key_c)
wiimote.nk_z = key(key_z)
wiimote.nk_stick_x = key_v,key_b
wiimote.nk_stick_y = key_n,key_m
```
Tha'ts the output:
```
sudo ./Dropbox/MoltenGamepad/MoltenGamepad/moltengamepad 
warning: Running MoltenGamepad as root is not recommended. Instead, please run as a non-superuser that has been granted the appropriate permissions.
loading /etc/xdg/moltengamepad/moltengamepad.cfg
driver: wiimote driver initialized.
Loading profiles from /home/gustavo/.config/moltengamepad/profiles/dota-2/dota2.2
parse: header is wiimote
parse: option "nk_ir_active" set.
parse: option "nk_accel_active" set.
parse: setting wiimote.nk_a = key(btn_mouse)
parse: setting wiimote.nk_b = key(btn_right)
parse: setting wiimote.nk_1 = key(key_8)
parse: setting wiimote.nk_2 = key(key_9)
parse: setting wiimote.nk_up = key(key_o)
parse: setting wiimote.nk_down = key(key_y)
parse: setting wiimote.nk_left = key(key_u)
parse: setting wiimote.nk_right = key(key_i)
parse: setting wiimote.nk_plus = key(key_leftalt)
parse: setting wiimote.nk_minus = key(key_7)
parse: setting wiimote.nk_home = key(key_h)
parse: setting wiimote.nk_ir_x = key(left_x+)
parse: setting wiimote.nk_ir_y = key(left_y+)
parse: setting wiimote.nk_c = key(key_c)
parse: setting wiimote.nk_z = key(key_z)
parse: setting wiimote.nk_stick_x = (key_v,key_b)
parse: setting wiimote.nk_stick_y = (key_n,key_m)
stdin: ready to read commands from standard input. Try "help" for more info.
```

jgeumlek, 2017-11-21:
I think I see the issue, but I don't have the time to really dig and test at the moment: that profile is generating `key_v` and `key_b` events, but they are still being sent to the virtual gamepad device rather than the keyboard device.

All the rest are wrapped in `key(...)` which redirects the translations to the virtual keyboard device instead.

Automagic shunting of keyboard events is on the horizon.

Unfortunately, I believe the `axis = (btn1,btn2)` syntax sugar is set to only work when not nested.

Try `wiimote.nk_stick_x = key(axis2btns(key_v,key_b))`.

Alternatively, you can make sure to assign the wiimote to the keyboard slot directly rather than `virtpad1` or whatever. If you use `set slots active_pads = 0`, MG should start assigning devices to the keyboard slot rather than the virtpads. You can then later increase the number of active pads to assign to later. (`active_pads` neither creates nor destroys virtual devices. It merely changes how MG allocates input sources)

gustavolinux, 2017-11-22:
`Try wiimote.nk_stick_x = key(axis2btns(key_v,key_b))`

That worked, thank you very much!!

closed 2017-11-22 by gustavolinux