jgeumlek/MoltenGamepad #53: map mini-keyboard with mouse to virtual gamepad

iRamos99, 2018-03-30:
I have a mini-keyboard with mouse pad (amazon link: https://www.amazon.com/gp/product/B071FT92F8/ref=oh_aui_detailpage_o02_s01?ie=UTF8&psc=1). I want to use this in my raspberry pi with retropie. The problem is that retropie's default mapping doesn't support the four keys on the top-right of the keyboard (volume up, volume down, homepage, and compose).

I ahve MoltenGamepad installed in my raspberry pi working with a wiimote. I would like to know if there is a way to make a gendevices .cfg file to map the keyboard and mouse events to a virtual gamepad.

Any help will be greatly appreciated.

Thanks!

jgeumlek, 2018-03-30:
Should be possible -- although don't get your hopes up with doing anything fancy with the touchpad.

Step one should be connecting the device and then using `evtest` in a terminal. Since it is a keyboard, you might need to run `evtest` as root in order to see this particular device.

This is the info you want to take note of:

* The name that the device presents for `evtest`.
* The vendor/product IDs, if available. `evtest` will print them at the top like `Input device ID: bus 0x3 vendor 0x413c product 0x2107 version 0x111`
* A list of all the events you care about, like "event the device emits -> event I'd like it to emit"

Once you have this info, you are halfway there. You should be good to base it off of [this gendev file,](https://github.com/jgeumlek/MG-Files/blob/master/gendevices/16-sony-standard.cfg) just substituting in the appropriate information.

---
Some considerations:

* Keyboards by default can only be read by root -- a mild protection against keyloggers. To use it in MG without giving MG root permissions, you'll likely need to add a udev rule.
* You should be fine with `exclusive = true` without needing `change_permissions`, since most games will ignore the original device as just a keyboard
* For the `[match line]` in the gendev file, you should be good with just `[name= ...]`, but if it is too generic you might want to add in the vendor/product stuff so that MG doesn't grab all your keyboards.

Finally, if you still want to use the keyboard part, MG unfortunately doesn't have a good way of selectively blocking/passing through the original events. Either set `exclusive=false` to allow all the original events to be emitted, or spend some tediously adding all the keyboard buttons to your gendev file and then telling MG to map those keys back to their standard keyboard events.

Another unfortunate missing function here is the ability to tell the gendev driver to temporarily set `exclusive` to false. Depending on your setup, this might call for a situation where you only launch MG when you launch your gaming interface. 

---
Feel free to ask for any further help, be it the gendev file writing, the udev stuff, or how to handle getting a good mapping in MG once MG can read the device.

iRamos99, 2018-03-31:
Thanks for pointing me in the right direction. This is what I have done so far:

- Created a file in the gendevices folder: "keyboard.cfg". (Sample below)
- Created profile: "h9_keyboard". (Sample below)

Results so far:

- MG can read the device and output as v_gamepad and v_keyboard events
- MG can't read the mouse events
- While on desktop the v_keyboard events are registered using evtest. But it doesn't work on applications like Mousepad or the console itself. It does work on the console when the desktop is not running.

Note:
This device shows as two separate devices in evtest, one for keyboard and one for mouse events. Both have the same product and vendor id's. I guess this is why MG can't read the mouse events.
Also, I'm having problems with multiple keys press at the same time. This could be the device itself.

I will keep trying different settings to see if  I can get MG to read the mouse events and to see if I can optimize the multiple key press issues.

Thanks for the help!

keyboard.cfg
```
["USB USB Device" vendor=0xc45 product=0x800a]

# Options
name = "h9keyboard"
devname = "kb_"
exclusive = "true"
change_permissions = "false"
flatten = "true"
rumble = "false"

# Event mapping
key_compose = "compose", "Compose face button"
key_homepage = "homepage", "Homepage face button"
key_volumeup = "volumeup", "V+ face button"
key_volumedown = "volumedown", "V- face button"
key_up = "up", "Up Arow"
key_down = "down", "Down Arow"
key_left = "left", "Left Arow"
key_right = "right", "Right Arow"
key_esc = "esc", "escape key"
key_1 = "1", "Number 1"
key_2 = "2", "Number 2"
key_3 = "3", "Number 3
key_4 = "4", "Number 4"
key_5 = "5", "Number 5"
key_6 = "6", "Number 6"
key_7 = "7", "Number 7"
key_8 = "8", "Number 8"
key_9 = "9", "Number 9"
key_0 = "0", "Number 0"
key_minus = "minus", "Key Minus"
key_equal = "equal", "Key Equal"
key_backspace = "backspace", "Key Backspace"
key_tab = "tab", "Key Tab"
key_q = "q", "Key Q"
btn_left = "mouse_leftbtn", "Mouse Left Button"
btn_right = "mouse_rightbtn", "Mouse Right Button"
rel_x = "mousepad_x", "MousePad X Axis"
rel_y = "mousepad_y", "MousePad Y Axis"
```

h9_keyboard
```
[h9keyboard]
compose = primary
volumedown = secondary
volumeup = third
homepage = fourth
space = start
leftalt = select
up = up
down = down
left = left
right = right
mouse_leftbtn = lt
mouse_rightbtn = rt
mousepad_x = left_x
mousepad_y = left_y
esc = key(key_esc)
1 = key(key_1)
2 = key(key_2)
3 = key(key_3)
4 = key(key_4)
5 = key(key_5)
6 = key(key_6)
7 = key(key_7)
8 = key(key_8)
9 = key(key_9)
0 = key(key_0)
minus = key(key_minus)
equal = key(key_equal)
backspace = key(key_backspace)
tab = key(key_tab)
q = key(key_q)
```

jgeumlek, 2018-04-01:
That looks good. Here's some thoughts about the issues you mention.

--- 

For the multiple presses, you should try to use `evtest` to diagnose whether you see the multiple presses on the original raw device node, or only on the MG v_device. You might want to launch MG temporally with `exclusive = false` in the gendev file. This'll let you open two `evtest` instances to watch both at once (otherwise MG will block the original events).

---

You are saying the v_keyboard events only show up when running in the raw console, but not when in a graphical session? That's weird. It might be an X thing. Do you see the MG virtual keyboard in the list when you run `xinput`?

If you are seeing the keyboard events get emitted on the v_keyboard, I don't currently have any ideas for why X might ignore them. The v_keyboard seems to be autodetected pretty smoothly over here.

---

The fact that the mouse events appears on a different device is a hurdle for MG. There is a way to try to get over it, but it limits you to only having one of these keyboard pads connected.

That way is to use the `flatten = true` feature to have MG flatten multiple devices into one. With this, just have two `[match lines]` in your gendev.

    ["USB USB Device" vendor=0xc45 product=0x800a]
    ["<mouse name>" vendor=0xc45 product=0x800a]
    
    ... the rest of the gendev file ...

This should recognize both devices, and flatten them into one input source for MG. If you had two of these keyboard things though, they'd be flattened down to just one as well!