jgeumlek/MoltenGamepad #55: Joy-con doesn't display.

fantoro, 2018-04-28:
Joycons aren't detected, I'm obviously using bluetooth, on my PC it says that it's connected but when I write print devices I get this:

```
print devices

(wiimote)
  no devices
(steamcontroller)
  no devices
(joycon)
  no devices
```

jgeumlek, 2018-07-02:
Hello. Sorry for the delay, it's been hectic on my end.

That's interesting. Can you verify that the relevant /dev/hidraw* nodes are being created when the device connects, and that your udev rules are altering them so that MG has the needed permissions?

It is possible that the joycon detection logic is incomplete.

fantoro, 2018-07-14:
Right now I'm not on my Linux machine so I can't do anything until the end of this month :P