jgeumlek/MoltenGamepad #56: Physical gamepades still visible, after installation as systemuser

Dogy007, 2018-06-29:
So i installed MoltenGamepad and used the installation script for "systemuser".
The service works fine , but the system still sees the physical controller.
My Setup:
Version:  retropie 4.3
Udev: 215 
controller: joycons
I followed the instructions to comment out the correct line in /lib/udev/rules.d/70-uaccess.rules
#SUBSYSTEM=="input", ENV{ID_INPUT_JOYSTICK}=="?*", TAG+="uaccess"
Also the compiled moltengamepad is placed in /usr/local/bin/moltengamepad.
I am wondering if i need to comment out some more lines in /liv/udev/rules.d ? 

Following files exist in this folder by default in retropie 4.3
10-local-rpi.rules                   
15-i2c-modprobe.rules                
15-rfkill.rules                      
40-usb_modeswitch.rules              
42-usb-hid-pm.rules                  
50-bluetooth-hci-auto-poweron.rules  
50-firmware.rules                    
50-udev-default.rules                
55-dm.rules                          
60-cdrom_id.rules                    
60-crda.rules                        
60-drm.rules                         
60-fuse.rules                        
60-gnupg.rules                       
60-inputattach.rules                 
60-joystick.rules                    
60-keyboard.rules                    
60-persistent-alsa.rules             
60-persistent-input.rules            
60-persistent-serial.rules           
60-persistent-storage-dm.rules       
60-persistent-storage.rules          
60-persistent-storage-tape.rules     
60-persistent-v4l.rules              
60-triggerhappy.rules
61-accelerometer.rules
61-partuuid.rules
64-btrfs.rules
69-libmtp.rules
69-libmtp.rules
70-power-switch.rules
70-uaccess.rules
71-seat.rules
73-idrac.rules
73-seat-late.rules
75-net-description.rules
75-persistent-net-generator.rules
75-probe_mtd.rules
75-tty-description.rules
78-sound-card.rules
80-drivers.rules
80-net-setup-link.rules
80-networking.rules
85-hwclock.rules
85-regulatory.rules
90-alsa-restore.rules
[rules.zip](https://github.com/jgeumlek/MoltenGamepad/files/2150942/rules.zip)

90-usbmount.rules
95-udev-late.rules
97-hid2hci.rules
99-systemd.rules
I added those rules as a zip folder, also willing to search these myself, if someone tells me what to look for ;)

When i try to pair my joycons as "pair" or single nothing happens, still the physical devices are detected.
I could also provide more informations , i would like to be helping with this issue where i can.

Dogy007, 2018-06-30:
I just tested with Retropie 4.4, which has udev version 232 and it works fine.

jgeumlek, 2018-07-02:
Hello. I'm currently doing a summer internship 8 time zones away from my home, so my time has been stretched a bit thin as of late. I also don't currently have access to a Pi.

Supporting versions of udev under 217 is always going to be a bit of a hack. Taking only the briefest of looks right now, the files in rules.zip appear to be fine:

* The relevant line in 70-uaccess.rules is commented out
* There isn't an obvious culprit of a later rule conflicting.

So I don't know for sure what is happening.

Some useful diagnostics:

- Do the permissions of the device nodes appear correct? With the udev rules in place, even without MG running, we should see the /dev/input/{event,js}* nodes corresponding to the physical controllers being assigned ownership to the gamepad account. I believe for joycon we also want the relevant /dev/hidraw* node.
- Is uaccess successfully being prevented? Try to open these nodes with an unpriviliged user account, and you should fail.
- Are the MG udev rules actually being triggered? Please check that there isn't a /etc/udev/rules.d/70-uaccess.rules file that is shadowing the one changed in /lib/. If you really want to delve into debugging the rules, `udevadm test <syspath>` is handy. Note that you need a syspath like `/sys/class/input/js0`, a device path like `/dev/input/js0` won't work with this command.