jgeumlek/MoltenGamepad #57: Crash on startup

Azelphur, 2018-09-14:
```$ gdb moltengamepad
GNU gdb (GDB) 8.2
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-pc-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from moltengamepad...(no debugging symbols found)...done.
(gdb) run
Starting program: /usr/bin/moltengamepad 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
loading /home/azelphur/.config/moltengamepad/moltengamepad.cfg

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7a3a794 in uw_update_context_1 (context=0x7fffffffe048, fs=0x7fffffffde88) at /build/gcc/src/gcc/libgcc/unwind-dw2.c:1395
1395    /build/gcc/src/gcc/libgcc/unwind-dw2.c: No such file or directory.
(gdb) bt full
#0  0x00007ffff7a3a794 in uw_update_context_1 (context=0x7fffffffe048, fs=0x7fffffffde88) at /build/gcc/src/gcc/libgcc/unwind-dw2.c:1395
        orig_context = {reg = {0x0, 0xbbec89db747cf400, 0x0, 0x7fffffffe048, 0x7fffffffde88, 0x7fffffffe0f0, 0x7fffffffe048, 0x8, 
            0x7ffff7a3b62a <_Unwind_Resume+58>, 0x7ffff7a3d476 <_Unwind_Find_FDE+102>, 0x7ffff7a3b62a <_Unwind_Resume+58>, 0x0, 0x0, 0x7fffffffddc0, 
            0x7ffff7a406c4, 0x0, 0x0, 0x7ffff7a406d5}, cfa = 0x7ffff7a406d5, ra = 0x7fffffffe048, lsda = 0x7ffff7a406c4, bases = {tbase = 0x7ffff7a3f1e0, 
            dbase = 0x7fffffffde88, func = 0x7ffff7a39ca0 <uw_frame_state_for+816>}, flags = 140737351474608, version = 93824993037192, 
          args_size = 93824992511466, by_value = "\te\327\367\377\177\000\000Pe\327\367\377\177\000\000`0"}
        cfa = <optimized out>
        i = <optimized out>
        tmp_sp = {ptr = 0, word = 0}
#1  0x00007ffff7a3ac57 in uw_init_context_1 (context=0x7fffffffe048, outer_cfa=0x7fffffffe278, outer_ra=0x55555556d24a)
    at /build/gcc/src/gcc/libgcc/unwind-dw2.c:1607
        ra = <optimized out>
        fs = {regs = {reg = {{loc = {reg = 18446744073709551544, offset = -72, 
                  exp = 0xffffffffffffffb8 <error: Cannot access memory at address 0xffffffffffffffb8>}, how = REG_SAVED_OFFSET}, {loc = {
                  reg = 18446744073709551552, offset = -64, exp = 0xffffffffffffffc0 <error: Cannot access memory at address 0xffffffffffffffc0>}, 
                how = REG_SAVED_OFFSET}, {loc = {reg = 0, offset = 0, exp = 0x0}, how = REG_UNSAVED}, {loc = {reg = 18446744073709551560, offset = -56, 
                  exp = 0xffffffffffffffc8 <error: Cannot access memory at address 0xffffffffffffffc8>}, how = REG_SAVED_OFFSET}, {loc = {reg = 0, 
                  offset = 0, exp = 0x0}, how = REG_UNSAVED}, {loc = {reg = 0, offset = 0, exp = 0x0}, how = REG_UNSAVED}, {loc = {
                  reg = 18446744073709551600, offset = -16, exp = 0xfffffffffffffff0 <error: Cannot access memory at address 0xfffffffffffffff0>}, 
                how = REG_SAVED_OFFSET}, {loc = {reg = 0, offset = 0, exp = 0x0}, how = REG_UNSAVED}, {loc = {reg = 0, offset = 0, exp = 0x0}, 
                how = REG_UNSAVED}, {loc = {reg = 0, offset = 0, exp = 0x0}, how = REG_UNSAVED}, {loc = {reg = 0, offset = 0, exp = 0x0}, 
                how = REG_UNSAVED}, {loc = {reg = 0, offset = 0, exp = 0x0}, how = REG_UNSAVED}, {loc = {reg = 18446744073709551568, offset = -48, 
                  exp = 0xffffffffffffffd0 <error: Cannot access memory at address 0xffffffffffffffd0>}, how = REG_SAVED_OFFSET}, {loc = {
                  reg = 18446744073709551576, offset = -40, exp = 0xffffffffffffffd8 <error: Cannot access memory at address 0xffffffffffffffd8>}, 
                how = REG_SAVED_OFFSET}, {loc = {reg = 18446744073709551584, offset = -32, 
                  exp = 0xffffffffffffffe0 <error: Cannot access memory at address 0xffffffffffffffe0>}, how = REG_SAVED_OFFSET}, {loc = {
                  reg = 18446744073709551592, offset = -24, exp = 0xffffffffffffffe8 <error: Cannot access memory at address 0xffffffffffffffe8>}, 
                how = REG_SAVED_OFFSET}, {loc = {reg = 18446744073709551608, offset = -8, 
                  exp = 0xfffffffffffffff8 <error: Cannot access memory at address 0xfffffffffffffff8>}, how = REG_SAVED_OFFSET}, {loc = {reg = 0, 
                  offset = 0, exp = 0x0}, how = REG_UNSAVED}}, prev = 0x0, cfa_offset = 0, cfa_reg = 7, cfa_exp = 0x0, cfa_how = CFA_REG_OFFSET}, 
          pc = 0x7ffff7a3b7b7 <_Unwind_Resume+455>, personality = 0x0, data_align = -8, code_align = 1, retaddr_column = 16, fde_encoding = 27 '\033', 
          lsda_encoding = 255 '\377', saw_z = 1 '\001', signal_frame = 0 '\000', eh_ptr = 0x0}
        sp_slot = {ptr = 140737488347768, word = 140737488347768}
        code = <optimized out>
#2  0x00007ffff7a3b62b in _Unwind_Resume (exc=0x5555555fddf0) at /build/gcc/src/gcc/libgcc/unwind.inc:235
        this_context = {reg = {0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x7fffffffde80, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0}, cfa = 0x0, 
          ra = 0x7ffff7a3b62b <_Unwind_Resume+59>, lsda = 0x0, bases = {tbase = 0x0, dbase = 0x0, func = 0x7ffff7a3b5f0 <_Unwind_Resume>}, 
          flags = 4611686018427387904, version = 0, args_size = 0, by_value = '\000' <repeats 17 times>}
        cur_context = {reg = {0x0, 0x3000000002, 0x0, 0x0, 0x6e0000005b, 0x0, 0x7c00000077, 0x7fffffffe180, 0xf, 0x8, 0x0, 0xffffffffffffff80, 0x8, 
            0x7fffffffe258, 0x7ffff78eb967 <malloc+151>, 0x1af, 0x8, 0x0}, cfa = 0x0, ra = 0x555555618a40, 
--Type <RET> for more, q to quit, c to continue without paging--
          lsda = 0x7ffff7c5b5fd <operator new(unsigned long)+29>, bases = {tbase = 0x555555618b18, dbase = 0x5555555a7a4b, func = 0x1e}, flags = 0, 
          version = 0, args_size = 22, by_value = "\036\000\000\000\000\000\000\000\020\213aUUU\000\000\260]"}
        code = <optimized out>
        frames = 4063479899412916480
#3  0x000055555556d24a in ?? ()
No symbol table info available.
#4  0x0000000000000001 in ?? ()
No symbol table info available.
#5  0x00007fffffffe320 in ?? ()
No symbol table info available.
#6  0x00007fffffffe4c0 in ?? ()
No symbol table info available.
#7  0x00007fffffffe460 in ?? ()
No symbol table info available.
#8  0x0000000000000001 in ?? ()
No symbol table info available.
#9  0x00007ffff7fa69e0 in ?? ()
No symbol table info available.
#10 0x00007fffffffe300 in ?? ()
No symbol table info available.
#11 0x00007fffffffe340 in ?? ()
No symbol table info available.
#12 0x0000000000000001 in ?? ()
No symbol table info available.
#13 0x00007ffff7fa64f0 in ?? ()
No symbol table info available.
#14 0x00007fffffffe2a0 in ?? ()
No symbol table info available.
#15 0x0000555555612b30 in ?? ()
No symbol table info available.
#16 0x0000000000000000 in ?? ()
No symbol table info available.```

Azelphur, 2018-09-14:
For reference, this occurs on archlinux x64, installed from AUR.

jgeumlek, 2018-12-06:
Hmm. I can't reproduce over here. Perhaps something with the config file? You can use `make clean` and `make debug` to build an executable with the symbol info to make that trace easier to understand.

note: I am not the owner of the AUR entry