jgeumlek/MoltenGamepad #59: Trigger command scripts with chords (new event translator)

mbenkmann, 2018-11-28:
Motivation: 
I have a 4 player game configured to use virtpadX for player X.
I start playing with 3 friends. We each press a button in turn on our physical controllers which causes the respective controller to be plugged into a virtpad slot.
Now the friend who is currently player 2 leaves (and takes his controller with him).
At this point we need to change our mapping from physical controllers to virtpads, so that we end up with virtpad1,...,virtpad3.

Currently I don't see how to achieve this with MoltenGamepad without getting up from the couch. What I would like to do is press Home+Start+Select on any controller to trigger the command "move all to nothing", thereby unplugging all controllers so that they will be replugged into the appropriate virtpad slots when we press buttons.

My idea to achieve this is to have an event translator

    script("commands for MG daemon")

to be used like this:

    gamepad.(start,select,home) = chord(unplug_all)
    gamepad.unplug_all = script("move all to nothing")

The script should be able to contain multiple commands separated by ";" or something like that.

Side note: I don't know if button releases are considered "notable input" for the purpose of plugging. If they are, there would be the need for a delay to allow the start, home and select buttons to be released before unplugging the controllers, to avoid replugging the controller used to send the command immediately. So the script would turn into something like

    gamepad.unplug_all = script("delay(500); move all to nothing")

mbenkmann, 2018-11-28:
Although not necessary to make the feature useful, it would be beneficial to be able to use pre-defined variables in scripts, such as $DEVICE to allow scripts like

    script("delay(500); move $DEVICE to nothing")

to allow a chord to just unplug the device the chord was input on.

jgeumlek, 2018-12-06:
This has been something I've considered for a while, in addition to allowing system shell commands as well (with some security considerations for the latter, of course).

The main roadblock is that these commands can take arbitrarily long amounts of time, i.e. `script("load profiles from /slow/network/fileshare")`, or in your examples, the new delay command. 

If we block to wait for the action to finish, any received inputs will be delayed. They should be buffered by the kernel for us, though, but that still isn't great. It also defeats the point of the delay.

If we don't block, I think we will want to still track asynchronously for the action to finish, otherwise multiple of these long-running actions might be fired simultaneously.

That isn't to say this is an impossible feature. Just that building an asynchronous event translator would take some work.

--------

Parser variables are feasible, but will complicate things a fair bit. Although this could be mitigated if only a handful of special well-known variables are permitted.

----------

As an alternative for the variable use case, another approach is to expose relevant device specific actions, much like how device specific options are exposed. Some made up examples:

* `remove_from_slot` for any device
* `toggle_motion_sensors`
* `recalibrate_sensors` for wiimote gyroscopes
* `activate_pairing` for joycons (switching from solo to paired, or vice versa)

And then make an event translator that can call these device specific actions on the device generating the event.

In perhaps a misguided decision long ago, I chose to have device options be preceded by `?` so that the parser can disambiguate them from device events. To not get into a huge mess of more inconsistency, these actions might want their own unambiguous prefix.

--------------------------

For the record, button release events are not considered notable.

mbenkmann, 2018-12-07:
Regarding script events and blocking: I would simply fork and send the commands over a socket connection. You already support listening for connections from clients. This way the new script() command does not introduce any issues that are not already present. A command line parameter --max-fork that limits the number of running child processes is probably advisable. If the limit is reached, further script calls would simply be discarded.