jgeumlek/MoltenGamepad #61: Allow Sysfs and patterned device matching for the generic devices

werpu, 2019-03-11:
Title says it all, atm the device mathiching is rather rigid, it only allows a fixed match however some ids might change over multiple machines or if you plug into another usb port.
A more flexible maybe regexped based device matching would allow to resolve those issues.
Also I noticed a sysfs match atm is not possible.

jgeumlek, 2019-03-11:
I think it would be great to expand the string-based matches with some sort of globbing/regexp matching, but I doubt I will get around to coding up the extra complexity in the gendev parsing/matching to handle that.

I'm not sure whether a full regexp match would be appropriate, or if a more restricted set of wild cards might be appropriate. With full regexp, now certain special characters like parentheses might start needing to be escaped...

What sysfs matches are you imagining? Are the sysfs paths stable across reboots? I'll admit it has been a while since I touched the udev monitoring code of MG.

werpu, 2019-03-12:
Well first of all thanks for answering, I know myself how hard it is to maintain an opensource project and having a professional and private life, so I do not take this as granted.

Well forget the sysfs idea... I think having a wildcard functionality suffices, but in the end it was a corner case I ran into which was not covered yet by the generic driver.

Just to explain why I had the idea of the sysfs and a small bug which I locally fixed yesterday.

I have a self made controller which will be the base of an arcade setup (maybe a cabinet in the long run). This controller has two analog arcade sticks two digital ones and about 22 buttons.
All is wired together with an arcade PCB and an internal usb hub.
I am combining this into 1-2 virtual joysticks depending on the settings (currently i am working on the q joystick gamepad like setup)
I ran into several issues which I could resolve (see my other already closed iusse #20)

https://github.com/jgeumlek/MoltenGamepad/issues/60

However I was programming the thing on a PC and plugged it now into an arcade setup and suddenly the analog sticks worked but the digital part did not work anymore (which worked flawlessly by using the phys address on the PC)

I thought it might be some setup problem that the pys is coming in slightly different as reported, usually you can resolve that rather easily by using wildcards and a substring. 
But after doing another debugging session in the code yesterday (have in mind I have not touched C++ in 25 years and never programmed in the apios linux exposes this takes a little bit, but the patterns are clear and concise so in the end no big deal just another language and api, same patterns I would use to pull this off), I found the reason. 

The reason why this happens is, that the device is not registering itself ans HID device, the phys address exposed might be an indicator

usb-xhci-hcd.2.auto-1.1.3/input2 xhci is human controller interface

So in the end the match function in the generic driver does not trigger

 struct udev_device* parent = udev_device_get_parent(dev);
    if (hidparent) {
        phys = udev_device_get_property_value(hidparent, "HID_PHYS");
        uniq = udev_device_get_property_value(hidparent, "HID_UNIQ");
    }

hidparent is reported back as null in my case, however you still can fetch the 
the phys address from the parent itself.

 if(parent && !match.phys.empty()) {
        phys = (char *) udev_device_get_property_value(parent, "PHYS");
        

reports back "\"usb-xhci-hcd.2.auto-1.1.3/input2\"" with quotes included

So when I extended the phys match to check for the string pattern (and ignoring the quotes)
via a strstr comparison and length check things started to work.
I will send in a patch in the next few days. I just want to check if there is a better way like doing a substring before comparisom (my c knowledge is rusty i am mostly java nowadays) before releasing my fix into the wild.

werpu, 2019-03-19:
I have created a pull request for the xhci issue https://github.com/jgeumlek/MoltenGamepad/pull/62

closed 2020-06-20 by jgeumlek