jgeumlek/MoltenGamepad #63: is it possible to map a gamepad button to a virtual mouse keypress

werpu, 2019-05-06:
Hello, I have about 99% of my setup working, but I am bumping my head now at a problem.
Moltengamepad seems to have somewhat of a virtual mouse support.
I have following situation, I have a generic joystick, which has exactly one button connected (emits btn_thumb atm) I try to map this button press into a mouse left click to recycle that button in an arcade setup with multiple mapped devices. For some reason I cannot remap the button. The mouse although it shows up as a device never emits the left_btn event in evtest.

`[name = "Ultimarc UltraStik Ultimarc Ultra-Stik Player 1" order="11"]
name = "mouse_1"
devname = "m1_"
exclusive = "false"
change_permissions = "false"
flatten = "false"
rumble = "false"
split = 1

btn_thumb = "btn_left", "Button Left"
`

The stick is registered into virtualjoypad1 i tried to move the device into another slot, but there seems to be no mouse slot only a keyboard one, which does not react either to any mapping atm.
Any ideas on what I am doing wrong here?

werpu, 2019-06-05:
Hi I just wanted to inform anybody, there is no need anymore to give an answer. I have programmed in the meanwhile my own mapper for my personal needs, which does exactly what I need. You might check out my just opened repo https://github.com/werpu/input_pipe .

closed 2019-06-05 by werpu

jgeumlek, 2019-06-12:
Glad you got your own thing working. MG started by scratching my own personal itches/grievances with other input mapping software.

For the record, it is indeed possible to use MG to map inputs to mouse clicks, and I use this frequently for some control mappings on a arcade machine as well.

If the input source is correctly generating mouse type events (such as `btn_left`), and the mouse type events are going to the `keyboard` slot, things should work.

You won't see the button clicks on the virtual mouse, they show up on the virtual keyboard. (Since the virtual keyboard should get detected as touchscreen-like absolute positioning cursor mover, this is not the strangest.)

If for some reason something expects the mouse clicks must come from the same device node that generates the `rel_x` and `rel_y` mouse movements, there is not currently a way to get MG to do that.

However, I have not run into any problems with the current way MG sends virtual mouse clicks. (Though I do admit the need to redirect things to the `keyboard` slot is a friction point)

werpu, 2019-06-13:
Hi thanks for the reply that resolves my original question. 

The reason why I started my own project was partially that in my configuration (3 devices in 4 different devices out) I hit a set of bugs, partially I was able to fix some of them partially not, due to my C having become somewhat rusty over decades of not using it. 

I must say I am pretty happy now with my choice of starting my own project. Python although this was my first project in this language turned out to be an excellent choice, they provide a reactive api for the event nodes and hence things despite being implemented in a rather slow language are super fast, and of course the maintenance becomes easier once you remove C from the equation (ymmv since you obviously know C better than me). My original plan was first try to get something up in python because there I could find all the bindings I needed upfront and then once the prototype works move to something faster and maybe write the bindings on top of C myself. It turned out that Python was more than fast enough thanks to the excellent libraries they provide (not the language itself per se)
All in all it took me 3 weeks of sparetime work to get my mapper up to a state where I could start to use it.

Despite not using MG anymore for the time being, thanks for your excellent work, it really was helpful and an inspiration for my own project (which is a little bit simpler because I don't cover that much ground)

jgeumlek, 2019-06-14:
Glad to hear it, it's nice to know it provided inspiration.

I never did really go around checking how slow other languages might be, or how garbage collection might show, especially on smaller computers or with more demanding games. But I'm not too surprised that Python works well.

The speed of C++ is one thing, but I also knew I'd probably be getting into the weeds of interacting with device drivers / ioctls, and I wanted this project to have minimal library dependencies.

True M:N mapping is a tricky business! Not just on the coding side, but the UI/configuration side. 

I also found that I was able to get a minimal version of MG working fairly quickly, with nicer features coming later. Getting device hiding working smoothly was pretty big obstacle.

Development on MG has slowed to a standstill not only because my life is busy, but also because the current dev version is essentially "good enough" for most of my needs now.

Best of luck, and have fun, with your input devices! : )