jgeumlek/MoltenGamepad #64: Uninstalling?

PeashyScript, 2019-08-17:
How do you uninstall it?

PeashyScript, 2019-08-17:
After I removed the binary and the conf files my gamepad isn't recognized at all

PeashyScript, 2019-08-17:
Well, I just had to reinstall my whole system to get my gamepads working again. Just a heads up: people normally include uninstall scripts with their software.

gustavolinux, 2019-08-17:
I'm using moltengamepad since 2016 and I should say it's a great peace of
software. Some lessons for you:

First, if you ask something, you should wait some time(not just 5 hours)
for an answer.
Second, you have to learn now to use Linux. Reinstall your operating system
is not a solution for a software issue.
Third, Don't blame a developer if you are the problem.

jgeumlek, 2019-08-21:
Hello. Sorry for taking a few days to get to this.

It sounds like you solved your problem, but I still feel like responding.

This software is still not particularly user friendly, and expects you to have some understanding on how the install works. We provide two different installs, and leave it to you to decide the trade-offs.

I'd be surprised if the single user install caused problems. When MG is not running, the devices should be visible like normal. And we intentionally do not ship a systemd service file, since this option assumes you want to be directly controlling the application.

The system user install is more involved, and we are explicit about it hiding devices even when MG is not running. 

---------------

However, I agree, we should have an uninstall script. It would be nice, and a good step for making this more readily consumable. Pull requests welcome.

In the mean time, the installation directory of this repo is very transparent. You can see exactly what files get placed by the install script. Every file additionally contains a comment at the top saying how it gets installed. And the relevant file paths are all near the beginning of the install script.

Just delete those files, and you should be good to go. (and likely, just removing the udev rules would suffice).

Sorry for whatever strife this software provided you.

HalfTough, 2022-05-21:
Just gonna add this for people who might get here by googling and still don't know how to get back to using controllers normally.
Two most important things to do are:

```
sudo systemctl disable moltengamepad  
sudo rm /etc/udev/rules.d/72-moltengamepad.rules
```

This will disable moltengamepad daemon, meaning it won't create virtual devices at start and remove udev rules, which will make regular devices available for normal user again.

zabinga73, 2023-07-26:
> Just gonna add this for people who might get here by googling and still don't know how to get back to using controllers normally. Two most important things to do are:
>
> ```
> sudo systemctl disable [...]

Thank you so much, im having the same issue and i was so scared this thread was gonna end with no more info than that guy reinstalling having worked. if its not in the readme it is indeed worth being there.

Mordius, 2024-01-28:
> Just gonna add this for people who might get here by googling and still don't know how to get back to using controllers normally. Two most important things to do are:
>
> ```
> sudo systemctl disable [...]

Well, same as the guy above, thank you so much!