jgeumlek/MoltenGamepad #66: displaying as multiple Gamepads (Wii Remote)

aviwad, 2019-09-14:
Hello! This program is amazing! However when I build it on Fedora 30 and run as root and connect my Wii Remote, it is seen as 4 Virtual Gamepads in the emulation software Gnome Games.
The first, third, and fourth do nothing, while all buttons are processed by the second. In this application, I cannot reorder players according to gamepads, so if these redundant "gamepads" under the same name could be removed, it'd be great.
[image]

Ryochan7, 2019-09-14:
You should be able to use the `--num-gamepads <n>` command line argument in order to change how many virtual gamepads are present.

jgeumlek, 2019-09-14:
Yes, the num gamepads option should give you what you want. The default is 4, and currently this number has to be given at start up.

I will also mention that MG allows you to "assign" your input devices to these various virtual gamepads, which allows you to dynamically re-order players even in applications that don't let you.

But if you are always playing single player, then setting the number of gamepads to 1 is a good choice.

aviwad, 2019-09-15:
OMG thank you both! This is amazing. Thank you so much!

closed 2019-09-15 by aviwad