jgeumlek/MoltenGamepad #67: permission problems

kroguli, 2019-12-10:
Hello
don't know, if this project is still alive...
If so:
while connecting my wiimote, MP writes:
open subdevice:: Permission denied
as well as:
device hiding: changing permissions of /dev/input/eventxx failed.

running moltengamepad as root does not show this behaviour, but I prefer not to do that.

I do have a group gamepad, of which I'm member.
What permissions do I have to change to get rid of these problems? 

thanks

jgeumlek, 2020-06-19:
Starting to work through a backlog of issues, apologies if this is no longer relevant.

However, the info here really isn't enough for me to diagnose anything.
When you see the message about eventXX hiding failing, take a look at the file ownership and file permissions of that eventXX node. 

To do the hiding, the MG process must be running as a user that _owns_ the device node.

Dgrif12, 2021-11-12:
[image]
I am having a similar issue, and it looks like, while moltengamepad owns the event files, it does not have access to them, and this is after I ran the modify_uaccess_rules.sh script.