jgeumlek/MoltenGamepad #70: Does DS4 touchpad clicking work? (and how)

Salvatos, 2020-02-20:
I've been trying to figure things out on my own but to no avail, so I would like to ask a few questions.

1. Can MoltenGamepad successfully send a touchpad click event from a Dualshock 4 or is that impossible due to how the kernel treats it as a separate device?

The older `12-playstation3+4-dt.cfg` had separate events for "home" (which I assume is the PS button) and "touchpad", but the newer `11-sony-standard.cfg` maps both to `key(316)`, which doesn't look encouraging.

2. Instead of installing and configuring MG, would it be possible to emulate a touchpad click with a simple keypress event via e.g. xmodmap? I couldn't figure out what MG actually does with a `key(316)` event but I imagine it has to be similar. Since the touchpad click is the only thing I am missing, this would be a more straightforward solution if I can just set a keyboard shortcut to send the right signal. However it doesn't seem possible to bind `BTN_xxxx` with xmodmap (which is what I think needs to happen?).

For context, my use case is running PlayStation Now via Lutris, which requires a DualShock 4. It works fine overall, but some games require the touchpad to open menus or perform actions and don't allow remapping the controls. I don't care whether I need to press the actual touchpad, an alternate button (Share doesn't seem to do anything on PS Now so it would be a good candidate) or a keyboard shortcut, but I need some way to activate that input and I'm not sure what the proper code for it is.

Thanks in advance for any help :)

TiZ-EX1, 2020-03-12:
> Can MoltenGamepad successfully send a touchpad click event from a Dualshock 4 or is that impossible due to how the kernel treats it as a separate device?

The problem is partially this but more that MoltenGamepad's resultant emulated devices are XBox 360 controllers that don't have the concept of a touchpad button. But the reason the touchpad is a separate button on DS4s is that the share button has OS-specific functionality. The touchpad does what the back/select buttons used to do. So I would recommend mapping select to PS Now's touchpad.

Salvatos, 2020-03-12:
Thank you for the advice. Unfortunately that doesn’t seem to work (but I could be doing it wrong). I have it mapped in Steam so that the right part acts as Start and the left as Select, and while Start opens/closes the in-game menu as expected, Select doesn’t seem to register at all.

TiZ-EX1, 2020-03-25:
It does sound like it's misconfigured. If you have parts of the touchpad mapped in Steam, that sounds like Steam Input is reading the raw HID device and doing its own thing, separately from MoltenGamepad. Steam Input and MG interfere with each other.

jgeumlek, 2020-06-19:
I think in between the last time I worked on MG and now, there have been some changes to how DS4s present themselves under linux. I'll need to explore that again.

At the time of writing the DS4 profiles, MG supported reading the touchpad click, but has not supported emitting a touchpad click.

(Ultimately, MG currently supports either emitting a generic gamepad, or a mimic of an xbox controller.)

I don't know much about PS Now, but can you generally configure it to use different key presses? The easiest should be something like what was described above: use a different button (like share) to emit some keycode (like btn_select), and then tell PS Now to treat btn_select events like the touchpad click.

Salvatos, 2020-06-19:
No, as far as I’m aware PS Now has no such configuration. The only option is language. I’m able to make the touchpad send _some other signal_ in various ways (Steam, xmodmap...), but I don’t know what signal PS Now actually expects from the touchpad, so I didn’t manage to work around this while I was subscribed. Fortunately, with Horizon: Zero Dawn coming to PC sometime soon, I’m not going to need PS Now again for the foreseeable future :)

jgeumlek, 2020-06-19:
Okay, so if it doesn't support config, thenI think it either:

- Reads the separate touchpad event device for touchad events (I don't think those separate device nodes existed back when I dealt with DS4s in MG!)
- Accesses the DS4 at a deeper hidraw level.

Although it might be feasible to get MG to create virtual touchpad nodes with a little work, that sounds messy.

I think if we wanted to do this right, we'd need to get MG to create virtual HID devices, and mimic the DS4 at that level. It would require getting deep into DS4's exact hid reports, but I think it is possible? A lot of work, but potentially a huge benefit to MG if we can get different output plugins working. We might even be able to support a virtual light bar this way...

-----------------

Looking at DS4s on linux now, I am a little amused: this project grew out of WiimoteGlue, designed to deal with the annoyances of Wiimotes splitting their events across multiple devices. That aspect has grown less relevant over time, as Wiimotes become less prevalent. But now DS4s are splitting their inputs across device nodes, requiring gluing them together once again...