jgeumlek/MoltenGamepad #72: Configure as virtual controller - potential vJoy alternative?

tdeeb, 2020-03-20:
I've been working on a robust [Twitch Plays bot](https://github.com/teamradish/TRTwitchPlaysBot), which utilizes [vJoy](https://github.com/shauleiz/vJoy), a virtual controller driver for Windows. The bot feeds inputs to vJoy to make it "press" buttons. For example, setting "button 0" causes it to press that button, and when mapped to an emulator or other game, essentially allows playing a game with it.

The bot itself runs on Linux, but since vJoy is Windows-only, inputs won't work on any other platform. To solve this, I've been looking for virtual controllers and came across MoltenGamepad. However, after reading through everything and giving it a brief test-run myself, I'm still not sure if it can act as a replacement to vJoy and want some advice since I'm not experienced with it.

In short, I want to know if MoltenGamepad can do the following:
* Configure up to 4 virtual controllers recognized by the OS
* Set up to 6 axes on each controller
* Set up to 32 buttons (or more) on each controller with numerical names (Ex. "Button 0")
* Handle pressing/releasing inputs through code and without an existing device

After all my searching, MoltenGamepad looks to be the most solid virtual controller I can find for Linux, and I greatly appreciate any help!

Ryochan7, 2020-03-20:
Not sure if you could make a software based system to handle that kind of mapping in MoltenGamepad. Your main requirements would likely be doable when using a physical device. You might have to look into making a small mapper program that takes advantage of uinput to create the virtual gamepad and configure it the way you want it; MoltenGamepad uses uinput for creating virtual devices and transmitting virtual events.

tdeeb, 2020-03-23:
I ended up taking your advice and writing a C program to create virtual gamepads in uinput, and everything is working great.

I searched for a software-based virtual controller on Linux for a long time with no luck, so here's [my simple implementation](https://github.com/teamradish/TRTwitchPlaysBot/blob/master/TRBot/TRBotCore/Native/SetupVController.c) to help anyone else in the future looking for this.

closed 2020-03-23 by tdeeb