jgeumlek/MoltenGamepad #74: Wiimote gyro movements send "invalid calculations?"

ryancordin, 2020-05-07:
I found a bug in 'wiigyromouse'  that is from the gyro data of the wiimote. Whenever I swing the wiimote quickly, there will be a pause to the cursor movement. This pause displays as "invalid calculations?." This specific output seems to be in regards to a 'deadzone', but even when setting a deadzone of 0, the result isn't changed. `wiimote.(nk_wm_accels, nk_wm_gyros, nk_a-) = wiigyromouse(deadzone=0)`

I believe this problem is related to not having a 'fast mode' correction to the gyro data. If I understand correctly, the wiimote gyros have two modes where sensor sensitivity will change depending on the speed of the wiimote. The first mode is the default one, in which normal, slow, consistent movements are read. The second mode is when the gyro data on the wiimote changes from higher speeds (such as when you make a sudden swing). I don't quite understand why there are two modes, but it's something that I've come across while using another project that utilizes wiimote data.

The other project is called FreePIE, and it is able to convert wiimote sensor data into useful button mapping, mouse inputs, or scripts. That program also has a problem with the 'fast mode', but instead of ignoring the 'fast mode', it assigns the input to the wrong axis readings. The result was unintentional spikes or dives to mouse movements (such as having a vertical dip while swinging the wiimote horizontally)

That programs 'fast mode' is located at the Github project 'FreePIE/FreePIE.Core.Plugins/Wiimote/DolphiimoteWiimoteData.cs' starting on line 137. It references wiic and wiibrew for the fast mode.

jgeumlek, 2020-06-19:
Oh, interesting. A lot of the motion stuff in MG is bare minimal get-something-working, but I was unaware of that fast mode.

Sounds like it might be a pain to support, especially depending on whether the kernel wiimote interfaces report it.

ryancordin, 2021-07-21:
I've been playing around with the program again, and it seems the 'invalid calculations' message is not what I thought. I assigned the gyro axis' individually to mouse(rel_*), and movement is quite good, albeit a bit too slow. I think the issue contained in my first post is that you require the accel data to be coupled with wiigyromouse. Is there a way to decouple this?

I actually have a need to ignore accel data completely, and to map the gyro data individually using the mouse(rel_*) binding, while also using the ratchet feature ("nk_a-"). The reason is that the assigned z-axis is actually my y-axis, and it's also inverted. the x-axis is correct. In addition, I cannot use my accel data as there is a lot of noise(treadmill). I would think something like this would work `wiimote.(nk_wm_gyro_z, nk_a-) = mouse(rel_y-)` but it doesn't. A bonus of using the mouse(rel_*) instead of wiigyromouse, is that I am can stay in virtpad1 and have the benefit of using controller buttons(joystick, chorded buttons, etc.) and also have the precision of mouse aiming.

ProtonAbyss, 2023-02-28:
I found out the solution to the problem. The code that is meant to capture invalid calculations in wiigyromouse.cpp is actually the cause for my issue. Specifically, the printf() function is the cause for the slowdown. If I comment out lines 213-216, there is no pause and the movements are consistent.

I highly suggest commenting out this code as it renders quick gyro-to-mouse movements impossible.