jgeumlek/MoltenGamepad #76: 4 point leds for wiimote irsensor

gustavoalara, 2020-06-06:
Hi,

I've found another interesting project for our wiimotes: xwiigun, https://github.com/hifi/xwiigun, it uses 4 point of ir LEDs instead of the simple top LEDs bar for better accuracy.

I tested it and works very well in my raspberry pi retropie based arcade cabinet.

Could be very cool if Molten would use this approach too

jgeumlek, 2020-06-19:
It would be nice, I just doubted how many people would go through the effort of finding 4 IR sources, rather than just using their pre-existing sensor bar.

We could have multiple code paths for processing IR data into mouse positions, and we could add an extra option to wiimote devices to switch between the different code paths.

gustavolinux, 2020-06-20:
I have used MoltenGamepad a lot for playinng Dota 2 using just one IR light source and worked very well.

benklop, 2020-07-13:
I'd be interested in this, as this is the LED configuration I currently have set up. I'm interested in the ability for this LED configuration to provide line-of-sight accuracy.