jgeumlek/MoltenGamepad #80: Segmentation fault with "make_keyboard = false"

FaithLV, 2020-08-13:
Adding `make_keyboard = false` to configuration causes a segfault on launch.

Installed from `master` and for single user.
Arch Linux 5.7.12-zen1-1-zen

```
(lldb) target create "moltengamepad"
Current executable set to 'moltengamepad' (x86_64).
(lldb) run
Process 101223 launched: '/usr/local/bin/moltengamepad' (x86_64)
loading /home/faith/.config/moltengamepad/moltengamepad.cfg
driver: sony driver initialized.
stdin: ready to read commands from standard input. Try "help" for more info.
Process 101223 stopped
* thread #1, name = 'moltengamepad', stop reason = signal SIGSEGV: invalid address (fault address: 0x118)
    frame #0: 0x00007ffff7f35d14 libpthread.so.0`__pthread_mutex_lock + 4
libpthread.so.0`__pthread_mutex_lock:
->  0x7ffff7f35d14 <+4>:  movl   0x10(%rdi), %eax
    0x7ffff7f35d17 <+7>:  movl   %eax, %edx
    0x7ffff7f35d19 <+9>:  andl   $0x17f, %edx              ; imm = 0x17F
    0x7ffff7f35d1f <+15>: nop
(lldb) bt
* thread #1, name = 'moltengamepad', stop reason = signal SIGSEGV: invalid address (fault address: 0x118)
  * frame #0: 0x00007ffff7f35d14 libpthread.so.0`__pthread_mutex_lock + 4
    frame #1: 0x00005555555767cf moltengamepad`__gthread_mutex_lock(pthread_mutex_t*) + 38
    frame #2: 0x0000555555576ac4 moltengamepad`std::mutex::lock() + 24
    frame #3: 0x0000555555578022 moltengamepad`std::lock_guard<std::mutex>::lock_guard(std::mutex&) + 42
    frame #4: 0x00005555555e6816 moltengamepad`virtual_device::check_delayed_events() + 56
    frame #5: 0x00005555555bbc6c moltengamepad`slot_manager::tick_all_slots() + 122
    frame #6: 0x0000555555567430 moltengamepad`main + 1209
    frame #7: 0x00007ffff7a46002 libc.so.6`__libc_start_main + 242
    frame #8: 0x0000555555566d3e moltengamepad`_start + 46
(lldb)
```

jgeumlek, 2020-08-15:
Should be fixed now. It seems when the fix in #78 did not make it onto master.

FaithLV, 2020-08-15:
Is fixed indeed, thanks!

closed 2020-08-15 by FaithLV