jgeumlek/MoltenGamepad #82: Rumble not working in wine

FaithLV, 2020-08-15:
Hi!

I am using a dualshock 4 with `mimic_xpad = true` and the inputs work very well, but rumble doesn't work in wine programs. It works when testing in native steam though.

I am not sure if it's a wine issue or moltengamepad and I have no clue on how to debug this.

Controller cfg: https://gist.github.com/FaithLV/d5591a8a61d9f8fd771a78753cbcb76c
moltengamepad.cfg: https://gist.github.com/FaithLV/e783974e5444c8e7dada6231a9beb4fd

clebercasali, 2020-08-15:
Run "wine control".
Go to "game controllers" - "test force feedback".
If it doesn't work, the console output could give a clue.

FaithLV, 2020-08-15:
[image]

The `Microsoft X-Box 360 pad` doesn't show under "Test Force Feedback".

clebercasali, 2020-08-15:
Does it appear in the "Joysticks" tab?
I can't remember where MoltenGamepad places the event node.
Is it /dev/input/eventXX or /dev/input/virtual/eventXX ?
Run evtest on it and check if it's reporting force feedback (evtest /path/to/eventxx).
You should see something like that:

> Event type 21 (EV_FF)
    Event code 80 (FF_RUMBLE)
    Event code 81 (FF_PERIODIC)
    Event code 88 (FF_SQUARE)
    Event code 89 (FF_TRIANGLE)
    Event code 90 (FF_SINE)
    Event code 96 (FF_GAIN)

FaithLV, 2020-08-15:
Yes, it does appear and also inputs display correctly.

[image]

The event was `/dev/input/event256`

```
Input device name: "Microsoft X-Box 360 pad"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 304 (BTN_SOUTH)
    Event code 305 (BTN_EAST)
    Event code 307 (BTN_NORTH)
    Event code 308 (BTN_WEST)
    Event code 310 (BTN_TL)
    Event code 311 (BTN_TR)
    Event code 314 (BTN_SELECT)
    Event code 315 (BTN_START)
    Event code 316 (BTN_MODE)
    Event code 317 (BTN_THUMBL)
    Event code 318 (BTN_THUMBR)
  Event type 3 (EV_ABS)
    Event code 0 (ABS_X)
    Event code 1 (ABS_Y)
    Event code 2 (ABS_Z)
    Event code 3 (ABS_RX)
  Event type 21 (EV_FF)
    Event code 80 (FF_RUMBLE)
```

clebercasali, 2020-08-15:
I see.
Wine can't access the rumble function from the "js" interface. It needs the "event" interface, but it didn't show up.
Maybe it's not looking up to event256.
Say your regular devices take up to event20.
Try to symlink event256 to event21.

FaithLV, 2020-08-15:
It suddenly started working, still doesn't display in `wine control joy.cpl` but works in-game!
Though what is strange is that `R3` does display when clicked, but does not get detected in-game. :+1: 

Will not close and not sure what fixed this.

clebercasali, 2020-08-15:
Weird. Wine must be getting some unexpected data from the MoltenGamepad event interface.

FaithLV, 2020-09-07:
Yeah, what is strange that FF worked only what one specific time 23 days ago. 🤣