jgeumlek/MoltenGamepad #88: Build failure

stephensrmmartin, 2020-10-23:
Two problems:

# The first problem: ldlibs missing
```
d source/plugin/wiimote; make -j3 --jobserver-auth=3,4 plug.a
make[1]: warning: -j3 forced in submake: resetting jobserver mode.
make[1]: Entering directory '/home/hwkiller/Builds/AUR/moltengamepad/src/MoltenGamepad-95b0d472770360bacbc4286fce13e0c1511ebbdb/source/plugin/wiimote'
mv -f .d/source/core/devices/generic/generic_file.Td .d/source/core/devices/generic/generic_file.d
cd source/plugin/steamcontroller; make -j3 --jobserver-auth=3,4 plug.a
make[1]: 'plug.a' is up to date.
make[1]: Leaving directory '/home/hwkiller/Builds/AUR/moltengamepad/src/MoltenGamepad-95b0d472770360bacbc4286fce13e0c1511ebbdb/source/plugin/wiimote'
cd source/plugin/joycon; make -j3 --jobserver-auth=3,4 plug.a
make[1]: warning: -j3 forced in submake: resetting jobserver mode.
make[1]: Entering directory '/home/hwkiller/Builds/AUR/moltengamepad/src/MoltenGamepad-95b0d472770360bacbc4286fce13e0c1511ebbdb/source/plugin/steamcontroller'
make[1]: warning: -j3 forced in submake: resetting jobserver mode.
make[1]: Entering directory '/home/hwkiller/Builds/AUR/moltengamepad/src/MoltenGamepad-95b0d472770360bacbc4286fce13e0c1511ebbdb/source/plugin/joycon'
make[1]: 'plug.a' is up to date.
make[1]: Leaving directory '/home/hwkiller/Builds/AUR/moltengamepad/src/MoltenGamepad-95b0d472770360bacbc4286fce13e0c1511ebbdb/source/plugin/steamcontroller'
make[1]: 'plug.a' is up to date.
make[1]: Leaving directory '/home/hwkiller/Builds/AUR/moltengamepad/src/MoltenGamepad-95b0d472770360bacbc4286fce13e0c1511ebbdb/source/plugin/joycon'
mv -f .d/source/core/protocols/socket/socket_out.Td .d/source/core/protocols/socket/socket_out.d
ar rcs source/core/mg_core.a source/core/commandline.o source/core/config_parse.o source/core/main.o source/core/moltengamepad.o source/core/options.o source/core/output_slot.o source/core/parser.o source/core/plugin_loader.o source/core/profile.o source/core/slot_manager.o source/core/udev.o source/core/uinput.o source/core/commands/assign.o source/core/commands/clear.o source/core/commands/load.o source/core/commands/move.o source/core/commands/print.o source/core/commands/save.o source/core/commands/set.o source/core/devices/device_manager.o source/core/devices/input_source.o source/core/eventlists/eventlist.o source/core/eventlists/gamepad_event_list.o source/core/protocols/debug_messages.o source/core/protocols/message_stream.o source/core/protocols/ostream_protocol.o source/core/virtual_devices/virtual_device.o source/core/virtual_devices/virtual_gamepad.o source/core/virtual_devices/virtual_kbm.o source/core/devices/generic/generic_device.o source/core/devices/generic/generic_file.o source/core/devices/generic/generic_manager.o source/core/devices/generic/generic_parse.o source/core/event_translators/axis/axis2axis.o source/core/event_translators/axis/axis2btns.o source/core/event_translators/axis/axis2rel.o source/core/event_translators/button/btn2axis.o source/core/event_translators/button/btn2btn.o source/core/event_translators/button/btn2rel.o source/core/event_translators/general/multitrans.o source/core/event_translators/general/redirect_trans.o source/core/event_translators/group/calibratable.o source/core/event_translators/group/exclusive_chord.o source/core/event_translators/group/simple_chord.o source/core/event_translators/group/stick_dpad.o source/core/event_translators/group/thumbstick.o source/core/event_translators/group/wiigyromouse.o source/core/protocols/socket/socket_in.o source/core/protocols/socket/socket_out.o
cat: source/plugin/wiimote/ldlibs: No such file or directory
```
This stems from source/plugin/wiimote/Makefile::`$(shell cat ldlibs)`
`ldlibs` does not exist.

After I just added a blank ldlibs, another error emerged.

# Second problem: Undefined references
```
/bin/ld: source/core/mg_core.a(eventlist.o): warning: relocation against `key_list' in read-only section `.text'
/bin/ld: source/core/mg_core.a(eventlist.o): in function `get_key_name(int)':
eventlist.cpp:(.text+0xb3): undefined reference to `key_list'
/bin/ld: eventlist.cpp:(.text+0xbf): undefined reference to `key_list'
/bin/ld: source/core/mg_core.a(eventlist.o): in function `get_axis_name(int)':
eventlist.cpp:(.text+0x113): undefined reference to `axis_list'
/bin/ld: eventlist.cpp:(.text+0x11f): undefined reference to `axis_list'
/bin/ld: source/core/mg_core.a(eventlist.o): in function `get_rel_name(int)':
eventlist.cpp:(.text+0x143): undefined reference to `rel_list'
/bin/ld: eventlist.cpp:(.text+0x14f): undefined reference to `rel_list'
/bin/ld: source/core/mg_core.a(eventlist.o): in function `get_key_id(char const*)':
eventlist.cpp:(.text+0x1c3): undefined reference to `key_list'
/bin/ld: eventlist.cpp:(.text+0x1ca): undefined reference to `key_list'
/bin/ld: source/core/mg_core.a(eventlist.o): in function `get_axis_id(char const*)':
eventlist.cpp:(.text+0x263): undefined reference to `axis_list'
/bin/ld: eventlist.cpp:(.text+0x26a): undefined reference to `axis_list'
/bin/ld: source/core/mg_core.a(eventlist.o): in function `get_rel_id(char const*)':
eventlist.cpp:(.text+0x2bc): undefined reference to `rel_list'
/bin/ld: eventlist.cpp:(.text+0x2c8): undefined reference to `rel_list'
/bin/ld: source/core/mg_core.a(eventlist.o): in function `lookup_event(char const*)':
eventlist.cpp:(.text+0x363): undefined reference to `key_list'
/bin/ld: eventlist.cpp:(.text+0x36a): undefined reference to `key_list'
/bin/ld: eventlist.cpp:(.text+0x3f3): undefined reference to `axis_list'
/bin/ld: eventlist.cpp:(.text+0x3fa): undefined reference to `axis_list'
/bin/ld: eventlist.cpp:(.text+0x42c): undefined reference to `rel_list'
/bin/ld: eventlist.cpp:(.text+0x433): undefined reference to `rel_list'
/bin/ld: warning: creating DT_TEXTREL in a PIE
collect2: error: ld returned 1 exit status
make: *** [Makefile:77: moltengamepad] Error 1
```

- [System] Arch linux 64

jgeumlek, 2020-10-23:
About ldlibs: that error should be nonfatal, and the build completed for me on a fresh clone. It is not ideal, and a blank file would do the correct thing, so lets go ahead and put those files for plugins that do not need to link any additional libraries.

As for the undefined references: it looks like something failed in producing key_list.o and the other _lists, or otherwise are not available during linking. Might there be an earlier error? Those key lists do tend to depend on what key codes you have defined in your kernel headers, but I would imagine the versions in the repo would suffice for Arch 64. I am on Arch 64, and the repo built fine on my end from a fresh clone.

Can you try `make clean; make eventlists;  make`?

stephensrmmartin, 2020-10-23:
Ah, that was it. make eventlists wasn't called. Thank you.

I may make a new AUR package for this, because it is currently missing some deps and makedeps; it was also last updated 2 years ago.

I just have one question about it - Is there a reason for trying to disable all other controllers and disabling uaccess? SC-controller and steam both use uniput and uaccess. Is it possible for me to just use a simplified udev rule where uaccess is used to manage the evdev nodes, and otherwise to create the permissions needed for the wii controllers and such? The reason is, that I have other controllers too (x360, steam controller) that I don't necessarily want to manage via MG. Ideally, MG would just handle the wiimotes as configured, and not touch other controllers.

We can close this issue thread and talk elsewhere if you would like. If I make an updated AUR package, I would like a way to ask you questions if possible.