jgeumlek/MoltenGamepad #102: How to properly turn a USB Keyboard (input) to Gamepad (output)

Khyze, 2021-08-21:
**This is only for playing videogames, so take your time to read and reply, isn't important.**

After 1 or 2 days I finally got it working (sadly, with sudo), with only two issues.

- First one is: I can't figure out how to do the Dpad (up, down, right, left, updown, leftright)
- Second one is: I managed to do the sticks with translators, not sure if is the reason on the "bug" of the movement, for example in a game I press "left", it shows feedback, I press "right" (without releasing "left") it shows feedback, but when I release "left" both seems to be "released" I can only test it with WINE (specially in the Windows version of PCSX2, which shows the outputs), but keys without translators seems to work perfectly, so I feel I made something wrong.

### This is my gendevices/test.cfg
```
["CHICONY` USB Keyboard"]
name = "test"
devname = "kb_"
exclusive = "true"
change_permissions="true"

#Start and Select L3 and R3
key_q = "start"
key_e = "select"
key_z = "thumbl"
key_c = "thumbr"

#Direccionales Izquierdo
key_w = "lup"
key_a = "lleft"
key_s = "ldown"
key_d = "lright"

#Direccionales Derecho
key_up = "rup"
key_left = "rleft"
key_down = "rdown"
key_right = "rright"

#Flechas
key_t = "up2"
key_f = "left2"
key_g = "down2"
key_h = "right2"

#ABXY
key_kp1 = "third"
key_kp2 = "first"
key_kp3 = "second"
key_kp5 = "fourth"

#Trigger and Bumpers
key_kp7 = "tl"
key_kp4 = "tl2"
key_kp9 = "tr"
key_kp6 = "tr2"
```

I mostly used the example code found in the documentation (there was a mistake, but I forgot where it was, I think one tr2 was listed as tl2 or something minor like that)

### This is my profiles/test.txt
```
[test]
test.lup = btn2axis(left_y,-1)
test.lleft = btn2axis(left_x,-1)
test.ldown = btn2axis(left_y,1)
test.lright = btn2axis(left_x,1)

test.rup = btn2axis(right_y,-1)
test.rleft = btn2axis(right_x,-1)
test.rdown = btn2axis(right_y,1)
test.rright = btn2axis(right_x,1)

test.up2 = btn2axis(abs_hat0y,-1)
test.left2 = btn2axis(abs_hat0x,-1)
test.down2 = btn2axis(abs_hat0y,1)
test.right2 = btn2axis(abs_hat0x,1)
```

moltengamepad.txt only loads that profile "load profiles from test.txt" which only has translators for stuff I couldn't make work otherwise. (I tried using "up", "left", "right" and "down" per button but didn't worked)

I wonder if WINE is the culprit, I do get this meaning the code is probably not wrong.
```
parse: setting test.up2 = updown-
parse: setting test.left2 = leftright-
parse: setting test.down2 = updown+
parse: setting test.right2 = leftright+
```

I also plugged another keyboard (same brand, the Amazon ones) and it works perfectly as I wanted (Player 2, but you know, with the same problems)

CommanderRag, 2022-03-06:
Did you find the solution?

Khyze, 2022-03-06:
> Did you find the solution?

Sorry, no, I got tired of tinkering around, sadly, I still need to switch to Windows and use Keyboard Splitter to play some games...

jgeumlek, 2022-04-28:
Hello. Sorry for a late reply.

WINE introduces a lot of complexity to this issue. We are going from Physical device -> Linux drivers -> MG -> virtual Linux input device -> WINE -> virtual Windows device -> Game. There's a lot of room for problems in that mapping, especially since anything that doesn't look exactly like a Xbox 360 pad has a big chance of making something somewhere confused.

Mapping a keyboard to an axis is tricky, as you noticed. MG is indeed dumb here,  where mapping two buttons to two extremes of an axis means key releases can clobber each other. (Press and hold left, output left. Press and hold right, output right. Release left, output a 0...) On arcade sticks, this isn't too big a problem, since a 4-way stick can't push right while still holding left.

To do this correctly, MG would need a group button translator to be written, that can watch both buttons and decide how to declobber them. Something like "test.(lleft,lright) = leftright". MG has some group translators written, but unfortunately a two-keys-to-an-axis is not one of them. Such a translator could better handle the virtual thumbsticks, as well as the dpad-as-an-axis.

Alternatively, an option could be added in virtual_gamepad.cpp to try to declobber there. Both approaches would still have edge cases if we have multiple inputs mapped onto the same virtpad's dpad.

About Dpads: this also relates to the above, since Xbox 360 pads decided to make the dpad into two axis events, making it impossible for a 360 pad to represent both left and right being pressed at the same time. You have the correct ABS_HAT there. If your problem is the clobbering issue, see above. If the issue is no events at all, I think something after MG is dropping them (WINE, or PCSX). 

If the issue is confusion over left, right, vs leftright-, leftright+:

* leftright is abs_hat0x, an x-axis for dpad. leftright- is the negative extreme of this axis, which would be pressing left. (positive is right)
* left is BTN_DPAD_LEFT, a digital button event supported by many controllers with drivers written with later drivers. MG will try its best to translate BTN_DPAD_LEFT and BTN_DPAD_RIGHT into ABS_HAT0X events if you ask MG to emulate a 360 pad.