jgeumlek/MoltenGamepad #104: unable to assign?

cantenna, 2022-04-28:
I have tried

assign slot virtpad2 to Name=SWITCH CO.,LTD. Controller (Dinput) ps4_dt_1

assign slot virtpad2 to uniq AZ-RB-N64P-215

assign slot virtpad2 to Uniq=AZ-RB-N64P-215 H0 Bus=0003 Vendor=2563 Product=0575 Version=0111 

can someone help with the correct syntax please?

jgeumlek, 2022-04-28:
Hello. Here's some explanation of the command. If you just want syntax, skip below the line. I'm going to type a lot here to act as rough bit of documentation for this mostly undocumented command.

The "assign" command is only needed in very specific situations, like when you have a physical device that should always be assigned to a specific virtpad, such as the "Player 2" section of an arcade cabinet might be always player 2. 

For most use cases, letting MG allocate controllers to slots when a button works better, as it avoids situations where the player 1 virtpad is claimed by some unused connected controller.

------
The syntax is "assign slot <slot_name> to <id_type> <id>". There is no support for a more complex matching of more than one id type.

The first line would need to be `assign slot virtpad2 to name ps4_dt_1`. This uses the MG name for a device. The underlying name "SWITCH CO.,LTD. Controller (Dinput)" of the source input device is not supported here.*

The second line seems fine to me, assuming that MG is correctly identifying that device as having the uniq string AZ-RB-N64P-215. "print devices ps4_dt_1" with the device connected should show that string. If it doesn't, this uniq string has been problematic in the past for MG, since bluetooth-vs-wired often changes the HID hierarchy MG tries to scan to find a uniq string.

The third line is unfortunately unsupported, the assign function does not work on more complicated matching.

-----

Why does it not match on the underlying source device names names? Because I imagined arcade systems where you might have multiple controllers of the same type, where it makes sense to force certain controllers to be certain virtpads. These devices would all have the same underlying name like "SWITCH CO.,LTD. Controller (Dinput)", so that would not work to distinguish them. But MG names can be reliably ordered if they were coming from a split (like one arcade keyboard device generating two virtual input sources), or could be easily fixed by disconnecting/reconnecting controllers in the appropriate order.

This also motivates the "uniq" and "phys" support: uniq would let a certain device be hardcoded, while phys could theoretically let a certain USB port be hardcoded.

IssaAboudi, 2022-09-07:
@jgeumlek What happens if pressing a button doesn't automatically assign the controller? That's the issue I am facing right now