jgeumlek/MoltenGamepad #106: Gyro as Right Stick or Mouse?

Fancy2209, 2022-05-21:
Hello, I was trying to make a Portal 2 Profile, for that I was trying to use motion+ gyro as a Stick or as the mouse, but neither seem to work, how am I supposed to set this right?

LinUwUxCat, 2022-11-08:
I have the exact same problem. i tried mapping the following:
```
wm1.(nk_ir_x,nk_ir_y) = stick(right_x,right_y)
wm1.(nk_wm_accel_x,nk_wm_accel_z) = stick(right_x,right_y)
wm1.(nk_wm_gyro_x,nk_wm_gyro_z) = stick(right_x,right_y)
wm1.(wm_gyro_x,wm_gyro_z) = stick(right_x,right_y)
wm1.(wm_ir_x,wm_ir_y) = stick(right_x,right_y)
```
but none work. I do have a working IR sensor bar, as well as a wiimotion plus, but even trying to bind the accelerator to the stick seems to be ignored.

LinUwUxCat, 2022-11-08:
Okay i found the (user) issue. There are some variables that need to be set in order to receive data - e.g. if you're playing with nunchuck and wiimotionplus and want to get gyro data, you need to set `wm1.?nk_gyro_active = true`. then bind `wm1.(nk_wm_gyro_x,nk_wm_gyro_y) = stick(right_x,right_y)`.

I don't know if this is still helpful to you but i noticed that it doesn't work quite well for portal 2, in fact IR reading works much better, so in that case it's 
```
wm1.?nk_ir_active = true
wm1.(nk_ir_x,nk_ir_y) = stick(right_x,right_y)
```

Fancy2209, 2022-11-10:
I don't have a USB Sensor Bar, only a wii one