jgeumlek/MoltenGamepad #111: Detect and Add Device to slot

IssaAboudi, 2022-09-07:
Hello all

I have a couple controllers but they aren't detected by moltengamepad and i'm not sure how to add them to the profiles loaded - I have [this](https://github.com/jgeumlek/MG-Files/blob/master/gendevices/10-xpad-dt.cfg) one added to my gendevices folder. I also created a cfg file in the root of ~/.config/moltengamepad called `moltengamepad.cfg` that looks like:

```
#useful settings
enumerate = true
num_gamepads = 2
profile_dir="~/.config/moltengamepad/profiles"
mimic_xpad = true
rumble = false
monitor = true 

#load preferred default mappings
load profiles from default_map
```
I don't see where to add a device and get that setup?

IssaAboudi, 2022-09-07:
I was following along with the readme here:  https://github.com/jgeumlek/MoltenGamepad

This part didn't work as expected - what is my next step?:
[image]

Triory, 2022-09-07:
Heads up: If any of this is obvious to you, I don't mean to offend you. I am definetely far from being an expert and am just trying to help.

Did you set it up as systemuser? If so, you should check the systemd log `systemctl status moltengamepad` to see if your config file is getting loaded properly. And whether the gamepad is even recognized. If not, you can try manually configuring the config path. Run `moltengamepad --help` to see the available launch options. Either edit `/etc/systemd/system/moltengamepad.service` or test just running it as root.

IssaAboudi, 2022-09-07:
@Triory Hey, no problem

So this is what my systemctl status looks like... This is probably the problem:

[image]

Any chance you know what I could do to fix this?

IssaAboudi, 2022-09-07:
> Did you set it up as systemuser? If so, you should check the systemd log `systemctl status moltengamepad` to see if your config file is getting loaded properly.

What does your config file look like? Maybe im not doing it right? (I shared my in the initial post)

> And whether the gamepad is even recognized. If not, you can try manually configuring the config path. Run `moltengamepad --help` to see the available launch options. Either edit `/etc/systemd/system/moltengamepad.service` or test just running it as root.

I'm not sure what to add/change? What would you do?

Triory, 2022-09-07:
It says `no moltengamepad.cfg found`.
Either:
1 - Pace your config file at `/etc/xdg/moltengamepad.cfg`

or

2 - Edit `/etc/systemd/system/moltengamepad.service` and at the end of the line `ExecStart=` add `--config-path /your/config/path`

IssaAboudi, 2022-09-07:
> It says `no moltengamepad.cfg found`.
> Either:
> 1 - Place your config file at `/etc/xdg/moltengamepad.cfg`

Done

> And afterwards restart the service by running `systemctl restart moltengamepad`

Done

It's still not detecting anything 😢

IssaAboudi, 2022-09-07:
Now I'm getting a `open subdevice::permission denied` 

[image]

Seems to have some relation to another issue; https://github.com/jgeumlek/MoltenGamepad/issues/67

Triory, 2022-09-07:
Maybe try running the `modify_uaccess_rules.sh` script again?

IssaAboudi, 2022-09-08:
> Maybe try running the `modify_uaccess_rules.sh` script again?

Still having my initial problem of no devices being detected.

IssaAboudi, 2022-09-08:
Interesting, there's no "virtual gamepad" listing in `evtest`, that's probably a big part of why this doesn't work..

[image]

Triory, 2022-09-09:
So what does the moltengamepad output say on start now?
I am not sure if you still need to specify a profile for mapping. You could try mapping manually to see if it works. 

If you run moltengamepad in console, you can give it instructions. (can also do it with the systemd service if you setup [moltengamepadctl](https://github.com/jgeumlek/moltengamepadctl)) Enter `help` and `print` to see options. You should be able to find the device name and the events it produces.

Then try running something along the lines of `xpad-dt.a = key(key_a)` where `xpad-dt` is your device and `.a` is the event. You should then be able to test the mapping by pressing the button producing the event, it should in this case press "a" on the virtual keyboard.