jgeumlek/MoltenGamepad #119: Anyone have this working on Fedora?

mak448a, 2023-07-05:
Does anyone have this tool working on Fedora? I'd like to use it to map my wiimote.

bekopharm, 2023-09-15:
I do. What's the problem?

mak448a, 2023-09-17:
I'll try again and tell you if there's any problem now

mak448a, 2023-09-17:
```
compiling source/core/commandline.cpp...
mv -f .d/source/core/commandline.Td .d/source/core/commandline.d
compiling source/core/config_parse.cpp...
mv -f .d/source/core/config_parse.Td .d/source/core/config_parse.d
compiling source/core/main.cpp...
mv -f .d/source/core/main.Td .d/source/core/main.d
compiling source/core/moltengamepad.cpp...
mv -f .d/source/core/moltengamepad.Td .d/source/core/moltengamepad.d
compiling source/core/options.cpp...
mv -f .d/source/core/options.Td .d/source/core/options.d
compiling source/core/output_slot.cpp...
mv -f .d/source/core/output_slot.Td .d/source/core/output_slot.d
compiling source/core/parser.cpp...
mv -f .d/source/core/parser.Td .d/source/core/parser.d
compiling source/core/plugin_loader.cpp...
mv -f .d/source/core/plugin_loader.Td .d/source/core/plugin_loader.d
compiling source/core/profile.cpp...
mv -f .d/source/core/profile.Td .d/source/core/profile.d
compiling source/core/slot_manager.cpp...
In file included from source/core/slot_manager.h:8,
                 from source/core/slot_manager.cpp:1:
source/core/uinput.h:34:33: error: ‘std::shared_ptr’ has not been declared
   34 |   int watch_for_ff(int fd, std::shared_ptr<virtual_device>& slot);
      |                                 ^~~~~~~~~~
source/core/uinput.h:34:43: error: expected ‘,’ or ‘...’ before ‘<’ token
   34 |   int watch_for_ff(int fd, std::shared_ptr<virtual_device>& slot);
      |                                           ^
source/core/uinput.h:40:22: error: ‘weak_ptr’ is not a member of ‘std’
   40 |   std::map<int, std::weak_ptr<virtual_device>> ff_slots;
      |                      ^~~~~~~~
source/core/uinput.h:16:1: note: ‘std::weak_ptr’ is defined in header ‘<memory>’; did you forget to ‘#include <memory>’?
   15 | #include <map>
  +++ |+#include <memory>
   16 | 
source/core/uinput.h:40:22: error: ‘weak_ptr’ is not a member of ‘std’
   40 |   std::map<int, std::weak_ptr<virtual_device>> ff_slots;
      |                      ^~~~~~~~
source/core/uinput.h:40:22: note: ‘std::weak_ptr’ is defined in header ‘<memory>’; did you forget to ‘#include <memory>’?
source/core/uinput.h:40:31: error: template argument 2 is invalid
   40 |   std::map<int, std::weak_ptr<virtual_device>> ff_slots;
      |                               ^~~~~~~~~~~~~~
source/core/uinput.h:40:31: error: template argument 4 is invalid
source/core/uinput.h:40:45: error: expected unqualified-id before ‘>’ token
   40 |   std::map<int, std::weak_ptr<virtual_device>> ff_slots;
      |                                             ^~
make: *** [Makefile:60: source/core/slot_manager.o] Error 1
```

qwertychouskie, 2023-09-17:
https://github.com/jgeumlek/MoltenGamepad/pull/99

KayJay7, 2023-09-17:
I'm also on fedora 38. After adding the missing `include` it still won't compile.

I get this error:

```
compiling source/core/udev.cpp...
In file included from /usr/include/c++/13/string:54,
                 from /usr/include/c++/13/bits/locale_classes.h:40,
                 from /usr/include/c++/13/bits/ios_base.h:41,
                 from /usr/include/c++/13/ios:44,
                 from /usr/include/c++/13/ostream:40,
                 from /usr/include/c++/13/iostream:41,
                 from source/core/udev.cpp:2:
/usr/include/c++/13/bits/basic_string.h:4402:12: error: partial specialization of ‘struct std::hash<std::__cxx11::basic_string<char, std::char_traits<char>, _Alloc> >’ after instantiation of ‘struct std::hash<std::__cxx11::basic_string<char> >’ [-fpermissive]
 4402 |     struct hash<basic_string<char, char_traits<char>, _Alloc>>
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/13/bits/basic_string.h:4441:21: error: specialization of ‘std::__is_fast_hash<std::hash<std::__cxx11::basic_string<char> > >’ after instantiation
 4441 |   template<> struct __is_fast_hash<hash<string>> : std::false_type { };
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/c++/13/bits/basic_string.h:4441:21: error: redefinition of ‘struct std::__is_fast_hash<std::hash<std::__cxx11::basic_string<char> > >’
In file included from /usr/include/c++/13/bits/stl_bvector.h:61,
                 from /usr/include/c++/13/vector:67,
                 from source/core/udev.h:4,
                 from source/core/udev.cpp:1:
/usr/include/c++/13/bits/functional_hash.h:295:12: note: previous definition of ‘struct std::__is_fast_hash<std::hash<std::__cxx11::basic_string<char> > >’
  295 |     struct __is_fast_hash : public std::true_type
      |            ^~~~~~~~~~~~~~
In file included from source/core/devices/../event_translators/../slot_manager.h:13,
                 from source/core/devices/../event_translators/../moltengamepad.h:12,
                 from source/core/devices/../event_translators/event_change.h:8,
                 from source/core/devices/device.h:11,
                 from source/core/udev.cpp:9:
source/core/devices/../event_translators/../options.h: In constructor ‘options::options()’:
source/core/devices/../event_translators/../options.h:17:13: error: use of deleted function ‘std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>::unordered_map() [with _Key = std::__cxx11::basic_string<char>; _Tp = option_info; _Hash = std::hash<std::__cxx11::basic_string<char> >; _Pred = std::equal_to<std::__cxx11::basic_string<char> >; _Alloc = std::allocator<std::pair<const std::__cxx11::basic_string<char>, option_info> >]’
   17 |   options() {};
      |             ^
In file included from /usr/include/c++/13/unordered_map:41,
                 from source/core/udev.h:7:
/usr/include/c++/13/bits/unordered_map.h:148:7: note: ‘std::unordered_map<_Key, _Tp, _Hash, _Pred, _Alloc>::unordered_map() [with _Key = std::__cxx11::basic_string<char>; _Tp = option_info; _Hash = std::hash<std::__cxx11::basic_string<char> >; _Pred = std::equal_to<std::__cxx11::basic_string<char> >; _Alloc = std::allocator<std::pair<const std::__cxx11::basic_string<char>, option_info> >]’ is implicitly deleted because the default definition would be ill-formed:
  148 |       unordered_map() = default;
      |       ^~~~~~~~~~~~~

...
```
there are many more `use of deleted function ...` but I guess that's just cascading from the first two `partial specialization` errors

mak448a, 2023-09-18:
> #99

Thanks! I'll try it out.

mak448a, 2023-09-18:
Unfortunately, another error happened even after the pull request

KayJay7, 2023-09-20:
I read issue #121 which doesn't look like the same error (the user didn't post the error messages), but there someone suggested a possible regression in gcc13 and said that it works with gcc12.

I tried compiling using a fedora 37 container using `toolbox create --image fedora-toolbox:37` and it works. On fedora it seems to be easier to [just use a container](https://fedoramagazine.org/a-quick-introduction-to-toolbox-on-fedora/) rather than installing a second toolchain. I choose fedora 37 because it should be "more similar" to fedora 38, and it shipped with gcc12, but you can probably use anything (container or not) with an older gcc.

mak448a, 2023-09-20:
Thanks! I'll try that. However, I ran into a few errors.
I fixed the first one by running `make eventlists`.
I fixed the second one by changing the line that starts with `go-md2man` to `echo`. Then everything compiled.
But now, there's a third error. After I run `installation/singleuser/configure.sh mak448a` and then `installation/singleuser/install.sh`, it throws
```shell
$ ./install.sh
Failed to send reload request: No such file or directory
```

TurtleIdiot, 2023-09-27:
This works when compiled using gcc 12 on arch!
Just do
`pacman -S gcc12`
and then run
`make CC=gcc-12 CPP=g++-12 CXX=g++-12 LD=g++-12`

arielenter, 2023-12-05:
> This works when compiled using gcc 12 on arch! Just do `pacman -S gcc12` and then run `make CC=gcc-12 CPP=g++-12 CXX=g++-12 LD=g++-12`

In ubuntu 23.10, I used 
`sudo apt install g++-12 gcc-12`
and
` make CC=gcc-12 CPP=g++-12 CXX=g++-12 LD=g++-12`

mak448a, 2024-05-01:
For anyone using Fedora, here's the dependencies:

`sudo dnf groupinstall "C Development Tools and Libraries"`
systemd-devel
golang-github-cpuguy83-md2man

6yntar05, 2024-10-14:
You can avoid compile errors with this diff:
```
diff --git a/Makefile b/Makefile
index 673c35e..37e445a 100644
--- a/Makefile
+++ b/Makefile
@@ -17,7 +17,7 @@ MG_BUILT_INS+=wiimote
 INPUT_HEADER:=""
 
 LDLIBS=-ludev -lpthread -ldl
-CPPFLAGS+=-std=c++14 -Wall -Wmissing-field-initializers -O2
+CPPFLAGS+=-std=c++20 -Wall -Wmissing-field-initializers -O2
 
 SRCS:=$(shell echo source/core/*.cpp source/core/*/*.cpp source/core/*/*/*.cpp)
 
diff --git a/source/core/uinput.h b/source/core/uinput.h
index 41183a1..035dd5c 100644
--- a/source/core/uinput.h
+++ b/source/core/uinput.h
@@ -9,6 +9,7 @@
 #include <string.h>
 #include <string>
 #include <vector>
+#include <memory>
 #include <mutex>
 #include <stdio.h>
 #include <thread>
```
Work with ArchLinux gcc (GCC) 14.2.1 20240910