jgeumlek/MoltenGamepad #124: error when compiling

TudorHH3000, 2023-09-10:
[fuck@fedora MoltenGamepad-1.2.1]$ make
compiling source/core/slot_manager.cpp...
In file included from source/core/slot_manager.h:8,
                 from source/core/slot_manager.cpp:1:
source/core/uinput.h:34:33: error: ‘std::shared_ptr’ has not been declared
   34 |   int watch_for_ff(int fd, std::shared_ptr<virtual_device>& slot);
      |                                 ^~~~~~~~~~
source/core/uinput.h:34:43: error: expected ‘,’ or ‘...’ before ‘<’ token
   34 |   int watch_for_ff(int fd, std::shared_ptr<virtual_device>& slot);
      |                                           ^
source/core/uinput.h:40:22: error: ‘weak_ptr’ is not a member of ‘std’
   40 |   std::map<int, std::weak_ptr<virtual_device>> ff_slots;
      |                      ^~~~~~~~
source/core/uinput.h:16:1: note: ‘std::weak_ptr’ is defined in header ‘<memory>’; did you forget to ‘#include <memory>’?
   15 | #include <map>
  +++ |+#include <memory>
   16 | 
source/core/uinput.h:40:22: error: ‘weak_ptr’ is not a member of ‘std’
   40 |   std::map<int, std::weak_ptr<virtual_device>> ff_slots;
      |                      ^~~~~~~~
source/core/uinput.h:40:22: note: ‘std::weak_ptr’ is defined in header ‘<memory>’; did you forget to ‘#include <memory>’?
source/core/uinput.h:40:31: error: template argument 2 is invalid
   40 |   std::map<int, std::weak_ptr<virtual_device>> ff_slots;
      |                               ^~~~~~~~~~~~~~
source/core/uinput.h:40:31: error: template argument 4 is invalid
source/core/uinput.h:40:45: error: expected unqualified-id before ‘>’ token
   40 |   std::map<int, std::weak_ptr<virtual_device>> ff_slots;
      |                                             ^~
make: *** [Makefile:60: source/core/slot_manager.o] Error 1
[fuck@fedora MoltenGamepad-1.2.1]$

xerxesbeat, 2023-09-22:
forgot to ‘#include <memory>' in source/core/uinput.h

BinaryQuantumSoul, 2024-07-23:
> forgot to ‘#include <memory>' in source/core/uinput.h

Thanks