jgeumlek/MoltenGamepad #128: Error 1 when compiling

analgaming12, 2024-02-26:
When I run "make" this is the output: 
`compiling source/core/commandline.cpp...
mv -f .d/source/core/commandline.Td .d/source/core/commandline.d
compiling source/core/config_parse.cpp...
mv -f .d/source/core/config_parse.Td .d/source/core/config_parse.d
compiling source/core/main.cpp...
mv -f .d/source/core/main.Td .d/source/core/main.d
compiling source/core/moltengamepad.cpp...
mv -f .d/source/core/moltengamepad.Td .d/source/core/moltengamepad.d
compiling source/core/options.cpp...
mv -f .d/source/core/options.Td .d/source/core/options.d
compiling source/core/output_slot.cpp...
mv -f .d/source/core/output_slot.Td .d/source/core/output_slot.d
compiling source/core/parser.cpp...
mv -f .d/source/core/parser.Td .d/source/core/parser.d
compiling source/core/plugin_loader.cpp...
mv -f .d/source/core/plugin_loader.Td .d/source/core/plugin_loader.d
compiling source/core/profile.cpp...
mv -f .d/source/core/profile.Td .d/source/core/profile.d
compiling source/core/slot_manager.cpp...
In file included from source/core/slot_manager.h:8,
                 from source/core/slot_manager.cpp:1:
source/core/uinput.h:34:33: error: 'std::shared_ptr' has not been declared
   34 |   int watch_for_ff(int fd, std::shared_ptr<virtual_device>& slot);
      |                                 ^~~~~~~~~~
source/core/uinput.h:34:43: error: expected ',' or '...' before '<' token
   34 |   int watch_for_ff(int fd, std::shared_ptr<virtual_device>& slot);
      |                                           ^
source/core/uinput.h:40:22: error: 'weak_ptr' is not a member of 'std'
   40 |   std::map<int, std::weak_ptr<virtual_device>> ff_slots;
      |                      ^~~~~~~~
source/core/uinput.h:16:1: note: 'std::weak_ptr' is defined in header '<memory>'; did you forget to '#include <memory>'?
   15 | #include <map>
  +++ |+#include <memory>
   16 | 
source/core/uinput.h:40:22: error: 'weak_ptr' is not a member of 'std'
   40 |   std::map<int, std::weak_ptr<virtual_device>> ff_slots;
      |                      ^~~~~~~~
source/core/uinput.h:40:22: note: 'std::weak_ptr' is defined in header '<memory>'; did you forget to '#include <memory>'?
source/core/uinput.h:40:31: error: template argument 2 is invalid
   40 |   std::map<int, std::weak_ptr<virtual_device>> ff_slots;
      |                               ^~~~~~~~~~~~~~
source/core/uinput.h:40:31: error: template argument 4 is invalid
source/core/uinput.h:40:45: error: expected unqualified-id before '>' token
   40 |   std::map<int, std::weak_ptr<virtual_device>> ff_slots;
      |                                             ^~
make: *** [Makefile:60: source/core/slot_manager.o] Error 1
`
I am running Arch Linux, on kernel 6.7.6-arch1-1. I've already updated everything. Thank you in advance for your time!

fedpo2, 2024-03-03:
im having the same problem

analgaming12, 2024-03-13:
> edit source/core/uinput.h and add under all the other #include
>
> that got it compiled for me, now how do I use it???

refer to the arch linux wiki